cocotb/cocotb #2139: Aldec Riviera-PRO stuck at matrix_multiplier examples

themperek, 2020-10-19:
CI seems to stuck at matrix_multiplier

```
# COUT:      -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:244  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
# COUT:      0.00ns INFO     cocotb                                      __init__.py:201  in _initialise_testbench           Running on Riviera-PRO version 2019.10.138.7537
# COUT:      0.00ns INFO     cocotb                                      __init__.py:208  in _initialise_testbench           Running tests with cocotb v1.5.0.dev0 from /home/az_devops_agent/agent/_work/2/s/cocotb
# COUT:      0.00ns INFO     cocotb                                      __init__.py:231  in _initialise_testbench           Seeding Python random module with supplied seed 123456789
# COUT:      0.00ns INFO     cocotb.regression                         regression.py:127  in __init__                        Found test test_matrix_multiplier.test_multiply_001
# COUT:      0.00ns INFO     cocotb.regression                         regression.py:467  in _start_test                     Running test 1/1: test_multiply_001
# COUT:      0.00ns INFO     cocotb.test._my_test.0x7f2f036ad390       decorators.py:312  in _advance                        Starting test: "test_multiply_001"
# COUT:                                                                                                                      Description: Automatically generated test
# COUT:                                                                                                                      
# COUT:                                                                                                                      	a_data: gen_a (Generate random matrix data for A)
# COUT:                                                                                                                      	b_data: gen_b (Generate random matrix data for B)
# COUT:                                                                                                                      
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 38902: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 38902: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 38902: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 38902: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 38902: The "9" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# PLI: PLI_ASSERT failed: File ./src/objs/pliobjects.cpp, Line 38902: The "15" property of "process" type objects is not implemented yet. Please contact Aldec Support to receive the latest status.
# COUT:      0.00ns WARNING  cocotb.gpi                                GpiCommon.cpp:431  in gpi_next                        Unable to create \0unnblk  via any registered implementation
# COUT:      0.00ns WARNING  cocotb.gpi                                GpiCommon.cpp:431  in gpi_next                        Unable to create  via any registered implementation
# COUT: /home/az_devops_agent/agent/_work/2/s/cocotb/scoreboard.py:76: DeprecationWarning: This Scoreboard implementation has been deprecated and will be removed soon.
# COUT: If this implementation works for you, copy the implementation into your project, while following cocotb's license agreement.
# COUT:   DeprecationWarning)
# COUT:      0.00ns INFO     cocotb.scoreboard.matrix_multiplier       scoreboard.py:225  in add_interface                   Created with reorder_depth 0
# PLI: $dumpvars(matrix_multiplier.sv:71): No access to Verilog memory, argument 2 will be skipped [systf:VCD0]
# PLI: $dumpvars(matrix_multiplier.sv:72): No access to Verilog memory, argument 2 will be skipped [systf:VCD0]
# PLI: $dumpvars(matrix_multiplier.sv:73): No access to Verilog memory, argument 2 will be skipped [systf:VCD0]
```

@garmin-mjames

imphil, 2020-10-19:
Also the CI run up to that point already takes surprisingly long, I didn't double-check what changed yet, but it seems like we're running more tests than before.

I'll also look into updating Riviera-PRO tonight to a newer version to see if the PLI_ASSERT lines are fixed.

garmin-mjames, 2020-10-19:
It may be related to #2079, based on the `gpi_next` failures. But then why did CI pass on that PR?

ktbarrett, 2020-10-20:
This occurs with Icarus as well. It appears to hang for a while before finishing during the matrix_multiplier.

ktbarrett, 2020-11-24:
It's not actually stuck, it just takes some time to finish. @garmin-mjames Should be contributing a PR soon that will reduce the amount of time it takes to run.

closed 2020-11-24 by ktbarrett

garmin-mjames, 2020-11-24:
This is the reason it takes such a long time:
```
# KERNEL: Warning: You are using the Riviera-PRO EDU Edition. The performance of simulation is reduced.
# KERNEL: Warning: Contact Aldec for available upgrade options - sales@aldec.com.
```
It takes about 430 seconds for the full test, which is much slower than all of the other simulators I tested, and this test alone takes up about half of the regression run time in CI.

imphil, 2020-11-24:
> It takes about 430 seconds for the full test,

Aldec donated an educational license of Riviera-Pro, which isn't representative of the speed you'd get from the real product.