cocotb/cocotb #2141: GPI: Address DEBUG log spam to cocotb.gpi logger

garmin-mjames, 2020-10-19:
There are two places that produce lots of DEBUG log statements in the GPI:

https://github.com/cocotb/cocotb/blob/a9cd7ad2625cd8cbfcde3fb5959fb7316b33b4a2/cocotb/share/lib/gpi/GpiCbHdl.cpp#L94-L100

https://github.com/cocotb/cocotb/blob/6eb637ae1ec4110fb3aa10ef04051fe8ef163108/cocotb/share/include/cocotb_utils.h#L57-L73

They are both related to tracing the program flow between GPI and Python.

I'd like to be able to filter out these messages without losing other DEBUG info on the `cocotb.gpi` logger. I would prefer to have them disabled by default with a way to enable them when debugging cocotb issues.

I don't know what the best approach is. New lower log level (#1226)? New distinct logger with a filter?

ktbarrett, 2020-10-20:
> New lower log level (#1226)?

Adding `TRACE` as a level, with the corresponding macro functions sound fairly easy to accomplish and conceptually covers both of those chatty cases.

> New distinct logger with a filter?

That's a good bit of work. If you were interested in adding more than one, I'd say build a facility so build more so we can have N. There really are only 4 important numbers: 0, 1, N, and 4 of course... Then we can call out different loggers in the `LOG_*` macros.

garmin-mjames, 2020-10-20:
@ktbarrett I was leaning towards `TRACE` as well.

`logging.TRACE = 5` ?

garmin-mjames, 2020-10-20:
Hmm, custom levels can clash with each other. The other approach is probably safer: https://docs.python.org/3.5/howto/logging.html#custom-levels

Thoughts @eric-wieser ?

PR #2502 (2021-04-18): Add custom TRACE logging level below DEBUG.
Closes #2141.

cocotb produces enormous amounts of `DEBUG` log messages relating to
the execution flow between Python and GPI. These messages bloat the
resulting logs and are not useful in most cases.

Move these messages to a new lower log level `TRACE`, which can be
enabled using `COCOTB_LOG_LEVEL=TRACE`.

TODO:
- [x] newsfragment
closed 2021-04-18 by marlonjames