cocotb/cocotb #2150: Infinite loop during design discovery when instant name and generate inside module have the same name

mkhazraee, 2020-10-21:
Hi,

I'm trying to make a checkpoint module to dump and load simulation state. I'm building my work based on this example that @themperek mentioned: https://github.com/themperek/apbi2c_cocotb_example/blob/parallel/tb/checkpoint.py

This is the base of my code, and then I use a collections.deque to iterate over all objects in a non-iterative way.

`    sims = set ()
    for sim in dut._handle.iterate(simulator.OBJECTS):
        sim_type  = sim.get_type()
        sim_name  = sim.get_name_string().split(".")[-1]  # this seems to be some bug
        # first layer full_name is same as sim_name
        if (sim_type in [simulator.REG, simulator.NET, simulator.MODULE, simulator.NETARRAY, simulator.GENARRAY]) and (sim_name not in sims):
            to_process.append((dut, sim, sim_type, sim_name, sim_name, ""))
            sims.add(sim_name)

    while (to_process):
        entity, sim, sim_type, sim_name, full_name, path = to_process.popleft()
        hdl = SimHandle(sim, entity._path + "." + sim_name)

        ... PROCESS REG/NET or ADD CHILDREN TO THE QUEUE 
            for child in hdl._handle.iterate(simulator.OBJECTS):
        ... `

I noticed in one of the modules the instant name had the same name as a generate block inside that module (inside the module that is instantiated). This caused the search process to get into a loop situation:
XXX.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs.csr_reg_nregs

and number of csr_reg_nregs increases in each iteration. The `entity._path + "." + sim_name` doesn't have the instant name inside the string, and generate blocks have a separate type (GENARRAY) where that has a child with the name of the generate. (I know it's super confusing, I was confused myself too.) I think that this is causing this issue. Not sure this is Icarus bug, or the way SimHandle works.

For now I changed the generate name to make things work, but I think it would be nice to find the bottom of this issue. Or if I should use other methods to do discovery please let me know. 

Moein

mkhazraee, 2020-10-21:
Actually another issue for the same code, if I dump all the registers and wires at the beginning of simulation I get no warnings and file size is 62.6MB. If I wait until some IP setup is complete, I get about 160 warnings and file size becomes 62.2MB. The interesting part is if I load the file and dump right then I get 62.6MB again! 

The warnings are one of these two:
VpiCbHdl.cpp:683  in VpiIterator, VPI: Implementation does not know how to iterate over vpiFunction(20)
GpiCommon.cpp:431  in gpi_next, Unable to create w_done via any registered implementation

Any insight where they are coming from and why they happen after some time is passed in the simulation?

Side note: I see there are other simulator types other than the 5 I'm using, based on https://docs.cocotb.org/en/stable/_modules/cocotb/handle.html 
or more complete one at 
https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/simulator/simulatormodule.cpp#L915

In the original example there wasn't even NET type. The other ones seem to be parameter and integer and the MEMORY one didn't have any hits in my design, as memory is seen as 2D net and captured by NETARRAY. My question is should I check for any other one for a full comprehensive module?

ktbarrett, 2020-10-21:
Could you create a minimum reproducible failing example (MRFE) for the recursive name bug and make that available to us?

ktbarrett, 2020-10-21:
> Any insight where they are coming from? or why file size changes? 

It's hard to say anything about what your code is doing without access to that code.

> VpiCbHdl.cpp:683  in VpiIterator, VPI: Implementation does not know how to iterate over vpiFunction(20)

I think this is pretty obvious. In your recursive decent you are not skipping functions and are trying to iterate over them, which makes no sense.

> GpiCommon.cpp:431  in gpi_next, Unable to create w_done via any registered implementation
> I figured the second one is always for always blocks with name, like always @ (*) begin: w_done. Is there a way to filter them out?

I'm not sure. The issue is that it cannot create an object for an always block. Seeing as an always block introduces a scope, it should be discoverable as a pseudo-region. The fact that it's not might be an Icarus or an VPI implementation bug. We will have to track that down. That should be a separate bug report and MRFE.

> My question is should I check for any other one for a full comprehensive module?

There are a lot of potential pitfalls when doing what you are trying to do. It assumes that simulator support is excellent and all stateful objects in a design can be discovered. Obviously it isn't taking into consideration things like functions. It doesn't appear to be taking into consideration constants/parameters. There is probably more. This kind of open-ended question belongs on the gitter and not in an issue.

This isn't the first time someone has tried to do this "save state and load it later" approach, and it has never worked or worked well from what I've seen. Since there is solid interest in this from multiple people, it might be beneficial to make this an open source cocotb extension. That way people can drive by and improve it over time, rather than people try to remake, poorly, over and over. Pinging @themperek whose example code inspired this.

themperek, 2020-10-21:
@ktbarrett Not sure if this is worth an extension but maybe an example? Only tested on simple designs.

This is originally made by @mciepluc

mkhazraee, 2020-10-22:
> I think this is pretty obvious. In your recursive decent you are not skipping functions and are trying to iterate over them, which makes no sense.

How can I distinguish a function? I'm checking for type and I'm only allowing 5 of them, but seems like that's not enough. And I didn't find any other method of the simulator to indicate something is a function. If the error is coming from the iterate itself, then how can I iterate and skip them? I saw you were using the same iterate in simulation.py file itself.

> I'm not sure. The issue is that it cannot create an object for an always block. Seeing as an always block introduces a scope, it should be discoverable as a pseudo-region. The fact that it's not [...]

I'll make an MRFE for this later today or tomorrow. 

> There are a lot of potential pitfalls when doing what you are trying to do. It assumes that simulator support is excellent and all stateful objects in a design can be discovered. From above, it isn't taking into consideration things like functions. It doesn't appear to be taking into consideration constants/parameters. There is probably more. This kind of open-ended question belongs on the gitter and not in an issue.

Well, if I'm simulating the same design and just resume it, the constants and parameters stay fixed. I looked into the simulator.py file and seemed like these are enough after I added NET type, but I was wondering if some are not included in python (simulator.py) but they are in VPI that I should consider. I'll ask it in the gitter. 

> This isn't the first time someone has tried to do this "save state and load it later" approach, and it has never worked or worked well from what I've seen. Since there is solid interest in this from multiple people, it might be beneficial to make this an open source cocotb extension. That way people can drive by and improve it over time, rather than people try to remake it, poorly, over and over. Pinging @themperek whose example code inspired this.

Agreed that this would be really helpful, specially for something like icarus that is slow. So I'm mostly wondering if we have enough control in cocotb right now to make this happen, and if so I'll try to make it as comprehensive as possible for now. One nice thing is I'm testing it on an open-source IP with lots of irregular code and if I make it work there it should possibly work for a good percentage of other designs too. But if not then it needs poking around with icarus too, specially that would benefit the most from this. 

I'm gonna make a MRFE for the looping problem in this issue, possibly later today or tomorrow.

mkhazraee, 2020-10-22:
Here is an MRFE based on the simple_dff example for the infinite loop:
[design_recovery_loop_MRFE.zip](https://github.com/cocotb/cocotb/files/5419365/design_recovery_loop_MRFE.zip)

ktbarrett, 2020-10-22:
I greatly simplified the test.

```python
import cocotb
import time

@cocotb.test()
async def test(dut):
    def wew(dut):
        print("traversing", dut._path)
        l = []
        for signal in dut:
            l.append(signal)
            print(signal._path)
        for sig in l:
            time.sleep(0.5)
            wew(sig)
    wew(dut)
```

This is pretty weird. It is finding `this_name` modules within the other_diff module. This points to an issue in Icarus.

```
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:893  in next_handle                     vpi_scan found 'dff.this_name.this_name'
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:266  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiModule(32)
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:420  in gpi_next                        Create a native handle
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:51   in check_and_store                 Checking dff.this_name.this_name exists
     0.00ns DEBUG    cocotb.this_name                              handle.py:95   in __init__                        Created
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:893  in next_handle                     vpi_scan found 'dff.this_name.this_name'
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:266  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiModule(32)
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:420  in gpi_next                        Create a native handle
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:51   in check_and_store                 Checking dff.this_name.this_name exists
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:58   in check_and_store                 Found duplicate dff.this_name.this_name
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:893  in next_handle                     vpi_scan found 'dff.this_name.this_name'
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:266  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiModule(32)
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:420  in gpi_next                        Create a native handle
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:51   in check_and_store                 Checking dff.this_name.this_name exists
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:58   in check_and_store                 Found duplicate dff.this_name.this_name
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:893  in next_handle                     vpi_scan found 'dff.this_name.this_name'
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:266  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiModule(32)
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:420  in gpi_next                        Create a native handle
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:51   in check_and_store                 Checking dff.this_name.this_name exists
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:58   in check_and_store                 Found duplicate dff.this_name.this_name
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:797  in next_handle                     End of type=92 iteration
```

mkhazraee, 2020-10-22:
Interesting, I just found out a way around it in my code, this works for the function one as well, not the always one. If I add this before iterating over children:

```
if len(entity._sub_handles)!=0:
    for child in hdl._handle.iterate(simulator.OBJECTS):
```

I understand how it fixes the function part, icarus is returning function as a module with no children. But no clue how it's fixing the naming part. To be honest I'm even not sure entity._sub_handles is the same as hdl._handle, as hdl is one of the objects in entity, but I guess since it's working sub_handle of father is handle of child! Something is not right there, but I think my current fix is reasonable as it's checking if there is any children iterate over them. 

Another weird situation was if there is a generate let's say with 12 iterations, it will find 12 submodule with same name of the generate name, and each of those submodules would have 12 submodules with the name generate_name[0..11]!  

I guess we should open an issue in icarus regarding this. You have a better understanding on the VPI side, but let me know if you want me to post it.

ktbarrett, 2020-10-22:
> How can I distinguish a function?

Actually after thinking about this a little more about this, we shouldn't be ever finding a function. Where it fails we call `vpi_get(vpiType, handle)`. Where handle was previously discovered as _not_ a `vpiFunction`. This implies that running that on the same handle returns a different result. This may be another bug.

mkhazraee, 2020-10-22:
Actually my fix is broken. When I tested it on the actual IP it stops at the very top level. Meaning it doesn't go deeper in the design as the generate block that has a wrapper of IP doesn't have any sub_handles!

garmin-mjames, 2020-10-22:
@ktbarrett 
> This is pretty weird. It is finding `this_name` modules within the other_diff module. This points to an issue in Icarus.

Looks like for some reason icarus is adding the **this_name** `vpiModule` handle 4 times instead of the 4 `vpiScope` handles `this_name[0]` through `this_name[3]`.

I added a call to `vpi_get_str(vpiFullName, new_hdl)` in `create_gpi_obj_from_handle` and it's returning `dff.this_name` for each one no matter how deep it goes:
```
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:893  in next_handle                     vpi_scan found 'dff.this_name.this_name.this_name.this_name'
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:266  in create_gpi_obj_from_handle      VPI: Created GPI object dff.this_name from type vpiModule(32)
```

Icarus doesn't have a proper **gen scope array** scope, only **gen scope**. I don't know if this is related to the apparent bug.

mkhazraee, 2020-10-22:
I've posted this on Icarus git as an issue, let's see if they can give more information or fix the bug. I'm gonna test it on ModelSim too. (https://github.com/steveicarus/iverilog/issues/380)

mkhazraee, 2020-10-23:
@ktbarrett @garmin-mjames any pointers on how to make the tester in pure C instead of python+C, as icarus folks asked? Something like which files I should include and potential high level functions to use, or where I can find the bare metal functions to build from. It's gonna take much less time to make a MRFE for them if you can give me some pointers. Thanks

ktbarrett, 2020-10-23:
Icarus ships with vpi_users.h which has all the VPI functionality. The Verilog standard (IEEE 1364-2005) goes over the design and usage of the VPI to some extent. You could also pick up "The Verilog PLI Handbook" by Sutherland which has more hands-on guidance.

Icarus might itself have something to help you write a VPI test. And likely have VPI self tests somewhere in the repo. I would ask them for assistance.

garmin-mjames, 2020-10-23:
I looked into this, and it's completely cocotb's fault:
https://github.com/cocotb/cocotb/blob/dc6ad51ebb387d36ed78389d7971717ab9935ec7/cocotb/share/lib/vpi/VpiImpl.cpp#L237-L248
We rely on the **gen scope array** name (**gen scope** prefix) not matching the parent name.
We essentially throw away the knowledge that we want a pseudo-region and attempt to recover it by only matching the names.

So a named generate `this_name` inside a module instantiation `this_name` breaks, and we create a `GPI_MODULE` instead of a `GPI_GENARRAY`.
We add this new `GPI_MODULE` handle as a sub-handle to itself. Then iteration over the sub-handle repeats this, forever.

mkhazraee, 2020-10-23:
I was able to make some tests in C, but seemed like iverilog doesn't have the loop at all, and then saw your message. 

For function it's also returning top0.this_name.f_foo is type vpiFunction and not vpiModule. 
For always block name it returns top0.this_name.my_name is type vpiNamedBegin, is this a bug or we can check for always in another method in cocotb?

Here's the test for ivtest, mainly found they have a recursive search example and I added my code and added some cases to their verilog. The only change to C is to print fullname instead of name, and added a case for vpiGenScope. 
[vpi.zip](https://github.com/cocotb/cocotb/files/5427509/vpi.zip)

Please let me know what you think about the other 2 cases, or the function one is related to the bug you found. Also is there a fix I can try?

cmarqu, 2020-10-23:
> You could also pick up "The Verilog PLI Handbook" by Sutherland which has more hands-on guidance.

The code examples of which can be downloaded at https://sutherland-hdl.com/books_and_guides.html#PLI%20Handbook

Donmar001, 2023-12-15:
@mkhazraee i have a netlist in the attachment,and i want to extract the logic cone by the PLI/VPI,how can i do that,and how to identify the DFF in the netlist,bro this is  my email donmarjr148@gmail.com,can you contact me?
[tmp.zip](https://github.com/cocotb/cocotb/files/13680484/tmp.zip)

cmarqu, 2023-12-15:
@Donmar001 Stop spamming people with unrelated things.