cocotb/cocotb #2152: always block name WARNING: Unable to create some_name via any registered implementation

mkhazraee, 2020-10-22:
I'm trying to make a checkpoint module to dump and load simulation state. Looping over simulator objects results in a WARNING if an always block has a name, like this:

```
always @(posedge clk) begin: some_name
   q <= d;
end  

```

WARNING: Unable to create some_name via any registered implementation

Reiterating what @ktbarrett said in #2150 : The issue is that it cannot create an object for an always block. Seeing as an always block introduces a scope, it should be discoverable as a pseudo-region. The fact that it's not might be an Icarus or an VPI implementation bug. We will have to track that down. That should be a separate bug report and MRFE.

Here is the MRFE:
[always_block_name_WARNING_MRFE.zip](https://github.com/cocotb/cocotb/files/5419388/always_block_name_WARNING_MRFE.zip)

garmin-mjames, 2020-10-22:
Icarus is returning `vpiNamedBegin` rather than `vpiAlways` (which is supported):
```
  5000.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:893  in next_handle                     vpi_scan found 'dff.this_name.some_name'
  5000.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:257  in create_gpi_obj_from_handle      VPI: Not able to map type vpiNamedBegin(33) to object.
  5000.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:426  in gpi_next                        Found a name but unable to create via native implementation, trying others
  5000.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:301  in __gpi_get_handle_by_name        Searching for some_name
  5000.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:308  in __gpi_get_handle_by_name        Skipping VPI implementation
  5000.00ns WARNING  cocotb.gpi                                GpiCommon.cpp:431  in gpi_next                        Unable to create some_name via any registered implementation
```

mkhazraee, 2020-10-22:
I've posted this on Icarus git as an issue, let's see if they can give more information or fix the bug. I'm gonna test it on ModelSim too. (https://github.com/steveicarus/iverilog/issues/380)

Donmar001, 2023-12-14:
@mkhazraee Asking help,How to identify the DFF in the code by the VPI,use the vpi_get_str to map the name?