cocotb/cocotb #2159: Keeping gui window open after test completes with Modelsim

mm4dd, 2020-10-26:


themperek, 2020-10-26:
I guessing you would need to remove/comment this line: https://github.com/cocotb/cocotb/blob/4d4fffaeedd3557601f706e1b5d872b15b0a7a42/cocotb/share/makefiles/simulators/Makefile.questa#L124

mm4dd, 2020-10-26:
Hi themperek,

thanks for the quick response. After trying your initial suggestion, I looked at the Makefile.questa and discovered the cocotb environment already sets up Modelsim correctly to use the waveform viewer in the gui. The problem turns out that the runsim.do script generated in the C:\cocotb\cocotb-1.4.0\examples\dff\tests\sim_build directory does not get over written for the gui settings if a runsim.do already exists in the directory. The first run of the test I ran the test without  GUI=1. This was the contents of the runsim.do file:

`# Autogenerated file
onerror {
	quit -f -code 1
}
if [file exists sim_build/work] {vdel -lib sim_build/work -all}
vlib sim_build/work
vmap -c
vmap work sim_build/work
vlog -work work +define+COCOTB_SIM -sv -timescale 1us/1us -mfcu +acc  C:/cocotb/cocotb-1.4.0/examples/dff/hdl/dff.v
vsim  -onfinish exit -pli C:/msys64/mingw32/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll   sim_build/work.dff
onbreak resume
run -all
quit`

After deleting the runsim.do file the file was regenerated when I used the GUI=1 switch

`# Autogenerated file
onerror {
	quit -f -code 1
}
if [file exists sim_build/work] {vdel -lib sim_build/work -all}
vlib sim_build/work
vmap -c
vmap work sim_build/work
vlog -work work +define+COCOTB_SIM -sv -timescale 1us/1us -mfcu +acc  C:/cocotb/cocotb-1.4.0/examples/dff/hdl/dff.v
vsim  -onfinish stop -pli C:/msys64/mingw32/lib/python3.8/site-packages/cocotb/libs/libcocotbvpi_modelsim.dll   sim_build/work.dff
add log -r *
`

Once the runsim.do file was regenerated with the correct settings the test opened modelsim in gui mode and I was able to load the runsim.do file and see all the signals in the waveform window.

Not sure if this should be logged as a bug or not?

Cheers
mmfdd

themperek, 2020-10-26:
>Not sure if this should be logged as a bug or not?

Hard to say, You should probably do `make clean` in those situations.

If your original issue is solved please close this issue. You may open a new one for discussion if `runsim.do` should regenerate every time.

imphil, 2020-10-26:
Whenever you change the command line arguments used with `make`, you need to call `make clean` first, as they do not automatically invalidate build artifacts.

mm4dd, 2020-10-26:
I will close the issue. Thanks for the help themperek and imphil.

closed 2020-10-26 by mm4dd