cocotb/cocotb #2173: Bit slices of binaryvalue are backwards

alexforencich, 2020-11-03:
cocotb bit indexes are reversed with respect to bit indexes in verilog - namely (bit_vector).value[0] returns the most significant bit instead of the least significant bit. 

Test verilog module:

    module test_mod (
        output wire [1:0] out
    );
    
    assign out[0] = 0;
    assign out[1] = 1;
    
    endmodule

Test python code:

    import cocotb
    from cocotb.triggers import Timer
    
    @cocotb.test()
    async def test_mod(dut):
        await Timer(2, units='ns')
        
        print(dut.out.value[0])
        print(dut.out.value[1])

Expected output:
    0
    1

Actual output:
    1
    0

alexforencich, 2020-11-03:
And this was marked as a question because....?

ktbarrett, 2020-11-03:
`BinaryValue` is all kinds of messed up, down to the design level, we have bug reports and PRs on everything wrong with it going back years. We intend to simply replace it soon #2059.

garmin-mjames, 2020-11-03:
`BinaryValue` is "big endian" (bits are ascending left-to-right) by default, so this happens. Almost all of the stuff around ascending vs descending is broken.

alexforencich, 2020-11-03:
In what alternate universe is the bit weight anything other than 2 to the power of the bit index?

eric-wieser, 2020-11-03:
The universe in which you mistakenly define "bit index" as "index in the string of big-endian ascii ones and zero". Obviously that universe is poorly though out, but its the one all cocotb code currently lives in...

alexforencich, 2020-11-03:
Alright, so it seems like this is not going to be fixed any time soon, so what's the simplest way - preferably something inline - to force a binaryvalue that has X in it into an integer?  I don't care if the Xs are replaced by 0s or 1s.

ktbarrett, 2020-11-03:
The [COCOTB_RESOLVE_X](https://docs.cocotb.org/en/stable/building.html#envvar-COCOTB_RESOLVE_X) variable.

alexforencich, 2020-11-03:
Great, that seems to be a reasonable workaround for the time being

ktbarrett, 2024-05-18:
BinaryValue is dead. LogicArray should not have this problem.

closed 2024-05-18 by ktbarrett