cocotb/cocotb #2174: Deprecate TestError/expect_error

cmarqu, 2020-11-03:
With the move to using assertions instead of raising exceptions, ``TestError``/``expect_error`` could/should be deprecated. https://github.com/cocotb/cocotb/pull/2170#discussion_r516712751

ktbarrett, 2020-11-03:
There is one reference that is located in the core libraries [here](https://github.com/cocotb/cocotb/blob/4d4fffaeedd3557601f706e1b5d872b15b0a7a42/cocotb/handle.py#L223), but I'm not sure why. When would iterating over a sub handles result in a `TestError`? I will check coverage to see if this is dead code.

garmin-mjames, 2020-11-03:
@ktbarrett Looks like it's catching this:
https://github.com/cocotb/cocotb/blob/4d4fffaeedd3557601f706e1b5d872b15b0a7a42/cocotb/handle.py#L959

ktbarrett, 2020-11-04:
Never trust the Github search feature. Lets me down every time.

PR #2177 (2020-11-08): Cleanup TestError
`TestError` was unnecessary here. It was used to fail the test if a signal could not be mapped to a Python SimHandle type. This could be easily done with a standard exception. This cleans up the last usage of `TestError` in core code and will allow for it's deprecation.

Closes #2174.
closed 2020-11-08 by ktbarrett