cocotb/cocotb #2185: simple_dff example (and probably others) broken for Verilator

toddstrader, 2020-11-10:
As discussed here:
https://gitter.im/librecores/cocotb-verilator?at=5fa165de74152347c210762b

simple_dff and other examples are not working with Verilator when using HEAD of both projects.
```
 22000.00ns ERROR    cocotb.regression                         regression.py:400  in _score_test                     Test Failed: test_dff_simple (result was AssertionError)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "/some/path/examples/simple_dff/test_dff.py", line 22, in test_dff_simple                                                                                                                                                                                                           
                                                                                                                         assert dut.q == val, "output q was incorrect on the {}th cycle".format(i)
                                                                                                                     AssertionError: output q was incorrect on the 1th cycle

```

The problem appears to be that since Verilator creates both port and local signal members for `clk`, `d` and `q`:
```
    // PORTS
    // The application code writes and reads these signals to
    // propagate new values into/out from the Verilated model.
    VL_IN8(clk,0,0);
    VL_IN8(d,0,0);
    VL_OUT8(q,0,0);
    
    // LOCAL SIGNALS
    // Internals; generally not touched by application code
    CData/*0:0*/ dff__DOT__clk;
    CData/*0:0*/ dff__DOT__d;
    CData/*0:0*/ dff__DOT__q;
```
and since the local input signals are always whacked by `eval()` since it always assumes you're talking to the PORTS members:
```
VL_INLINE_OPT void Vtop::_combo__TOP__1(Vtop__Syms* __restrict vlSymsp) {
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop::_combo__TOP__1\n"); );
    Vtop* const __restrict vlTOPp VL_ATTR_UNUSED = vlSymsp->TOPp;
    // Body
    vlTOPp->dff__DOT__clk = vlTOPp->clk;
    vlTOPp->dff__DOT__d = vlTOPp->d;
}
```
and since cocotb is getting VPI handles to the LOCAL SIGNALS:
```
     0.00ns DEBUG    cocotb.dff                                    handle.py:218  in _discover_all                   Discovering all on dff
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:686  in VpiIterator                     vpi_iterate type=36 returned NULL
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:686  in VpiIterator                     vpi_iterate type=114 returned NULL
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:701  in VpiIterator                     Created iterator working from 'dff' with type vpiModule(32)
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:893  in next_handle                     vpi_scan found 'dff.clk'
     0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:287  in initialise                      VPI: clk initialized with 1 elements
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:267  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiReg(48)
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:420  in gpi_next                        Create a native handle
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:51   in check_and_store                 Checking dff.clk exists
     0.00ns DEBUG    cocotb.clk                                    handle.py:95   in __init__                        Created
```
the values that cocotb attempts to set via VPI are constantly overridden.  In fact when you open the VCD from this all signals are always at zero.

I'm sure this isn't the correct solution, but this can be further demonstrated by this patch which causes the test to pass:
```
$ git diff examples/simple_dff/dff.sv
diff --git a/examples/simple_dff/dff.sv b/examples/simple_dff/dff.sv
index 442389f7..eba30a60 100644
--- a/examples/simple_dff/dff.sv
+++ b/examples/simple_dff/dff.sv
@@ -3,7 +3,13 @@
 
 `timescale 1us/1us
 
-module dff (
+module dff;
+    logic clk, d, q;
+
+    dff_dut dff_dut (.*);
+endmodule
+
+module dff_dut (
   input logic clk, d,
   output logic q
 );
```

This removes the PORTS members in the verilated class and stops the always-update-inputs behavior.

It was my understanding that this worked at one point, however I'm not sure what has changed.

cocotb: 4edd09dc093b90846c3fe1fced7dba936d6c3f55
Verilator: 5097698a0614db81bc24f73ad95c4ab005ce10f2
Python 3.6.5
Locally built and pip installed

garmin-mjames, 2020-11-10:
I've been investigating the wrapper port vs module port issue while working on updating the main verilator loop used by cocotb (#2105).

I've been able to determine that using iteration from the module handle returns handles to the internal signals (`dff.clk`), while doing a `vpi_handle_by_name('dff.clk', NULL)` lookup will return handles to the wrapper signals (`TOP.clk`). This has to do with using the special case of an empty string name for the wrapper.

Unfortunately, when trying to fix a different issue with icarus, I created a problem with verilator in https://github.com/cocotb/cocotb/commit/7351ccf3fe345d60f5158979d03aa7e670064e52. The fix for icarus was changed in https://github.com/cocotb/cocotb/commit/4d4fffaeedd3557601f706e1b5d872b15b0a7a42, which should fix the **simple_dff** example with verilator.

The iteration vs name lookup issue still needs to be resolved, so avoid iteration on a module handle in verilator for now.

ktbarrett, 2020-11-30:
Just tested the simple_dff example with Verilator 4.105 devel rev v4.104-4-ge74cc32f and it passes. If you are still having issues, reopen the issue.

closed 2020-11-30 by ktbarrett