cocotb/cocotb #2187: can't get signal under generate block with vcs, using systemVerilog

YangLeee, 2020-11-11:
when I want to capture a signal under generate block,like this:
in sv file:
`for(i=0; i<2; i=i+1) begin : PACKET
my _module module1(...)`

in cocotb test:
topmodule.PACKET[0].module1

I run make:
`make SIM=vcs`
it runs wrong and log:
File "/usr/local/lib64/python3.6/site-packages/cocotb/handle.py", line 306, in __getattr__
raise AttributeError("%s contains no object named %s" % (self._name, name))
 AttributeError: topmodule contains no object named PACKET
so, how to use signal under generate block with vcs?

cocotb environment:
[image]
simulator and verison:
vcs Version O-2018.09-SP
python version:
3.6.8

ktbarrett, 2020-11-11:
I'm fairly certain this has been fixed on master in #2143. Could you try [installing the development version](https://docs.cocotb.org/en/stable/install_devel.html#install-devel) and re-running your test? Pinging @garmin-mjames.

cmarqu, 2020-11-11:
Unfortunately, I don't believe it's fixed in master, I still see the problem with VCS in ``test_array`` with ``TESTCASE=test_discover_all``. I'll try to give more information soon.

garmin-mjames, 2020-11-11:
The workaround in #2143 is only compiled for Icarus. 
If `vpi_handle_by_name` doesn't resolve the `vpiGenScopeArray` handle, the iteration over parent workaround may work.
You could try adding a check for VCS to:
https://github.com/cocotb/cocotb/blob/9df43e5c0a797299ff5abc994da7cbbb31756e5e/cocotb/share/lib/vpi/VpiImpl.cpp#L305

cmarqu, 2020-11-11:
@garmin-mjames That workaround doesn't seem to work for VCS, it directly finds a ``vpiReg``. Here is a part of the log comparing Xcelium with VCS (I added the "Returning handle..." locally).

make SIM={xcelium,vcs} COCOTB_LOG_LEVEL=DEBUG TESTCASE=test_discover_all

Xcelium, works:

```
    10.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:893  in next_handle                     vpi_scan found 'array_module.asc_gen'
    10.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:243  in create_gpi_obj_from_handle      Found pseudo-region array_module.asc_gen (hdl_name=array_module but name=asc_gen)
    10.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:268  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiModule(32)
    10.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:421  in gpi_next                        Create a native handle
    10.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:51   in check_and_store                 Checking array_module.asc_gen exists
    10.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:56   in check_and_store                 Returning handle asc_gen
```

VCS, doesn't work:
```
    10.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:893  in next_handle                     vpi_scan found 'array_module.asc_gen[16].sig'
    10.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:287  in initialise                      VPI: asc_gen[16].sig initialized with 1 elements
    10.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:268  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiReg(48)
    10.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:421  in gpi_next                        Create a native handle
    10.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:51   in check_and_store                 Checking array_module.asc_gen[16].sig exists
    10.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:56   in check_and_store                 Returning handle asc_gen[16].sig
```

garmin-mjames, 2020-11-11:
Is VCS able to get and scan a `type=92` iterator from array_module? That's `vpiInternalScope`, where generate arrays should show up.

cmarqu, 2020-11-11:
No, it says
```
    10.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:797  in next_handle                     End of type=44 iteration
    10.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:799  in next_handle                     No valid type=92 iterator
    10.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:818  in next_handle                     No more children, all relationships tested
    10.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:442  in gpi_next                        Reached end of iterator
```

garmin-mjames, 2020-11-11:
@YangLeee Can you run your test with `COCOTB_LOG_LEVEL=DEBUG` and post the log lines leading up to the `AttributeError`?

garmin-mjames, 2020-11-11:
_( Documenting for myself and other maintainers )_
From investigating with @cmarqu, it looks like VCS doesn't properly support `vpiInternalScope` and instead the objects under the generate blocks are found by iterating on the parent module handle.

A consequence of this hierarchy flattening is putting everything as sub-handles on the module handle, along with name collisions due to:
https://github.com/cocotb/cocotb/blob/9df43e5c0a797299ff5abc994da7cbbb31756e5e/cocotb/handle.py#L241-L243
Where we throw away anything before the last `.`

---

Regarding this issue directly, there's a workaround to get a handle to the object under a generate block:

```python3
topmodule._id('PACKET[0].module1', extended=False)
```

@YangLeee can you try using that to get a handle to `module1`?

YangLeee, 2020-11-12:
> _( Documenting for myself and other maintainers )_
> From investigating with @cmarqu, it looks like VCS doesn't properly support `vpiInternalScope` and instead the objects under the generate blocks are found by iterating on the parent module handle.
> 
> A consequence of this hierarchy flattening is putting everything as sub-handles on the module handle, along with name collisions due to:
> https://github.com/cocotb/cocotb/blob/9df43e5c0a797299ff5abc994da7cbbb31756e5e/cocotb/handle.py#L241-L243
> 
> 
> Where we throw away anything before the last `.`
> Regarding this issue directly, there's a workaround to get a handle to the object under a generate block:
> 
> ```python
> topmodule._id('PACKET[0].module1', extended=False)
> ```
> 
> @YangLeee can you try using that to get a handle to `module1`?

@garmin-mjames  oh, when I use:
```python
topmodule._id('PACKET[0].module1', extended=False)
```
it works! I can get the handle to `module1`
thanks very much.

YangLeee, 2020-11-12:
@garmin-mjames 
here is the information before I use the method you pose before,hope to give more details.
when I still use `topmodule.PACKET[0].module1` and run my test with `COCOTB_LOG_LEVEL=DEBUG`:
```

  2504.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:911  in next_handle                     vpi_scan found 'topmodule'
  2504.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:245  in create_gpi_obj_from_handle      VPI: Created GPI object from type vpiModule(32)
  2504.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:420  in gpi_next                        Create a native handle
  2504.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:51   in check_and_store                 Checking topmodule exists
  2504.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:58   in check_and_store                 Found duplicate topmodule
  2504.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:815  in next_handle                     End of type=92 iteration
  2504.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:836  in next_handle                     No more children, all relationships tested
  2504.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:441  in gpi_next                        Reached end of iterator
```
and the wrong Info 
```
File "/usr/local/lib64/python3.6/site-packages/cocotb/handle.py", line 306, in __getattr__
raise AttributeError("%s contains no object named %s" % (self._name, name))
 AttributeError: topmodule contains no object named PACKET
```
notice that it's supposed to get `PACKET[0]` , but actually get`PACKET`?

garmin-mjames, 2020-11-12:
@YangLeee I'm glad that worked for you.

closed 2020-11-12 by garmin-mjames

garmin-mjames, 2020-11-12:
If you have further issues feel free to re-open this issue or create a new one.

jennyzhao007, 2024-06-09:
> @garmin-mjames here is the information before I use the method you pose before,hope to give more details. when I still use `topmodule.PACKET[0].module1` and run my test with [...]

Have you resolve this issue? I have the same errors. When use _id to get the Inst‘s object, but no signals in it.
attr of iface_port_tx_inst is ['__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattr__', '__getattribute__', '__getitem__', '__getstate__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__iter__', '__le__', '__len__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__setitem__', '__sizeof__', '__str__', '__subclasshook__', '__weakref__', '_child_path', '_compat_mapping', '_def_file', '_def_name', '_discover_all', '_discovered', '_fullname', '_handle', '_invalid_sub_handles', '_len', '_log', '_name', '_path', '_sub_handle_key', '_sub_handles', '_type', 'get_definition_file', 'get_definition_name']