cocotb/cocotb #2189: Agree on and add clang-format configuration

imphil, 2020-11-11:
Our C/C++ code is currently a mix of styles and not auto-formatted. I'm proposing to add a clang-format configuration to solve that.

In addition, we should agree on a C/C++ coding style. I tried to play around with clang-format to get it to match what we currently have, but our style is so inconsistent that it will be a rather large diff anyways. I'd therefore go with a full reformat with a standard coding style already supported in clang-format.

As usual, this is a matter of taste, but also doesn't matter much. I'd go with the Google C/C++ coding style (https://google.github.io/styleguide/cppguide.html), which is rather comprehensive and well supported in clang-format. (It has a couple things in there which I personally find odd, but that's expected in any style and I ultimately don't care enough.)

- [ ] Agree on a style
- [ ] Add a clang-format config file to the tree
- [ ] Reformat all C/C++ code with clang-format.
- [ ] Add clang-format lint to CI.

ktbarrett, 2020-11-11:
Need to take #1148 into consideration. It had to do with the `Python.h` include being moved around, but I'm afraid I forget the details.

ktbarrett, 2020-11-25:
Apparently that's not an issue, I just tried it. I'm fine with the Google code style. [clang-format-lint-action](https://github.com/DoozyX/clang-format-lint-action) can be used for linting.

garmin-mjames, 2020-12-09:
Google style is fine with me. We may be able to use [include-what-you-use-action](https://github.com/emilgedda/include-what-you-use-action) for include checking.

PR #2429 (2021-03-07): Add clang-format
Closes #2189.

Add Google clang-format file with slight modification for smaller diff.

Apply to C++ files. I left the VPI/VHPI standard headers alone and exclude them from the CI linting.
It's quite a large diff. I looked over the output and only cleaned up the alignment on comments and
where string literals were broken across lines.

Add clang-format linting, and also exclude the examples directory.

The format moved the includes around, and so on MacOS `vpi_user.h` didn't have the typedef for `uint64_t`. I added `__APPLE__` to the `stdint.h` include for both VPI and VHPI he [... cut]
closed 2021-03-07 by marlonjames