cocotb/cocotb #2195: Add functions to GPI for determining which procedural interfaces are implemented

ktbarrett, 2020-11-15:
The GPI abstracts away the knowledge of the procedural interface actually used (VPI, VHPI, or FLI), but one disadvantage is it precludes the user from adding new interface specific code. Currently you would have to add a new feature to the GPI abstracting the feature for all interfaces. This isn't always possible or desirable. For example, in the case of VPI's system tasks. Making a GPI-level abstraction for registering new system tasks is not really possible. Instead users should be able to write interface specific code, but will need someway to know whether it's safe to load those extensions/call those functions.

To remedy this limitation new functions should be added to the GPI to allow the user to determine which procedural interfaces are available. For example `gpi_vpi_available()`, `gpi_vhpi_available()` and `gpi_fli_available()`.

garmin-mjames, 2020-11-18:
So an example would be to check `gpi_vpi_available()` first, then directly interact with the VPI?

Would there be any use cases for a VPI/VHPI/FLI only feature that could be exposed out of the GPI, or should we continue to restrict the interface to the `gpi_*` functions.

ktbarrett, 2020-11-18:
> So an example would be to check gpi_vpi_available() first, then directly interact with the VPI?

That's correct. 

> Would there be any use cases for a VPI/VHPI/FLI only feature that could be exposed out of the GPI

In this case we either have the user define the system task and call the VPI functions after checking `gpi_vpi_available()`, or defining the system task and calling `gpi_vpi_register_task(....)` which would call the vpi functions or return an error code if it isn't available?

garmin-mjames, 2020-11-18:
The second part was more of a general question about how strict of an abstraction layer GPI should be over the individual interfaces.

There are some features that are only supported by a subset, like the VPI system tasks, or for VHPI/FLI there's driver creation. Does non-universal functionality belong in the GPI, and how should it be exposed to GPI users if so.