cocotb/cocotb #2214: Documentation: Improve Documentation for Power users Compilation & Elaboration

jshaker000, 2020-11-22:
My team is interested in using cocotb for cosimulation, but we have our own custom make flow with Questa and Xcelium. It would be helpful to discuss where the VPI needs to be included in the flow (we found that it seems to be needed in elaboration). Also, I see for xcelium in that makefile you include options for 'access' and so on.

Discussing for power users how to configure cocotb for different Vendors in a multistage flow (with seperate compilation / elaboration) would be helpful in case we want to modify our flow to use cocotb rather than fit in the given makefiles. Also a discussion of how the environment variable 'MODULE' is used and how that essentially springboards to starting your python bench would help. We figured some of this out, but it might be pretty hacky and fragile, so some official documentation on it would be appreciated.

Thanks!

garmin-mjames, 2020-11-22:
Thanks for opening an issue and providing feedback!

For now, see #1718 for an example using a custom Tcl flow.

cmarqu, 2020-11-22:
I guess such a document should start out as a wiki page, maybe as a sister page to https://github.com/cocotb/cocotb/wiki/cocotb-Internals

ktbarrett, 2020-11-22:
We have talked before about documenting how to use cocotb without the makefiles so users can use their own automation tools. I think that should go in RTD.

jshaker000, 2020-11-22:
As an end user, I'd ideally like to see a section on this in the wiki, perhaps as its own page under references, or under Simulator Supporr or Build Options (these were the places that we originally looked).

Ad I had said, I got something working in Xcelium by copying over all of the flags in elaboration that I didnt recognize, but that included linking the VPI vut also thinga like access crw, (also soke option that I think allowed debug access)

 which Im not sure is actually needed by cocotb and soon.

So a discussion of what is actually going on would be greatly helpful. This would also better put my mind at ease when cocotb updates and I can see womehwere what options I need to change (if any), rather than parsing through the provided makediles again.

I also didnt see an enviornment variable for specifying paths to verilog include files, which need to be listed to the compilation by path but not by file name. The point being is that we have custom flow that handles lots of these things, and bolting on some options needed by cocotb onto our flow is much more convenient than any other option.

Thanks!

themperek, 2020-11-22:
I have added a simple/minimum example for Xcelium in https://github.com/cocotb/cocotb/issues/1718

ktbarrett, 2020-11-22:
@jshaker000 cocotb is fundamentally a VPI/VHPI/FLI cosimulation library. You will see an option to the final simulator call like `-pli /???/???/libcocotbvpi_ius.so`. That line is what is actually loading cocotb. The only required environment variable is `MODULE` which specifies a Python module to load cocotb test definitions from.

The "Building Variables" documentation could better be split into "makefile" things and "cocotb" things so it's clear for users who don't use the makefiles what is important to consider.

jshaker000, 2021-01-15:
I'd like to highlight in the docs a few things for power users trying to integrate cocotb into their flow, rather than using the default make file
1)  Only the environment variable "MODULE" needs to be set, and this variable is used to pick the python module that has the tests
2) In Xcelium:
Elaboration options:
-loadvpi (In your python lib) cocotb/libs/libcocotbvpi_ius:vlog_startup_routines_bootstrap
Perhaps coco-config can point to where your python library is and where the VPI is?
-createdebugdb
This was in the Makefile but I'm not exactly sure if this is needed(?)
Access
-access +rwc
It seems like Xcelium by default checks if VPI is loaded and automatically makes access to rwc on every net, so this argument seems a bit redundant

In Questa:
It seems like these options are needed
Simulation:
-pli (cocotb path) cocotb/libs/libcocotbvpi_modelsim.so
Again, perhaps cocotb-config can pass this path
-onfinish exit

So Xcelium needs many options for elaboration, wheras Questa needs options for simulation but not elaboration. This kind of info is very helpful for custom flow.

cmarqu, 2021-01-15:
https://github.com/cocotb/cocotb/pull/2340/files should cover most of this, it would be nice if you could review.

IIRC I also tried Xcelium without the `-access +rwc` option and I found it was needed - maybe it's different between versions.
`-createdebugdb` is supposedly using a faster internal database format, but only supported in Xcelium, not in Incisive, so `Makefile.ius` doesn't have it.

The equivalent of `-onfinish exit` I would expect is already in existing user scripts.

jshaker000, 2021-01-16:
Everything there is correct, but it discusses single stage flow (compilation + elaboration together). I think its worth mentioning for multi-stage flows which arguments are needed at compiletime vs runtime. (which I had mentioned above).

cmarqu, 2021-01-16:
Ah yes, I'll amend.

cmarqu, 2021-02-22:
@jshaker000 BTW, if you want to find the sub-stage(s) for a certain xrun option, you can run `xrun -helpall -helpusage | grep <myoption>` which will contain e.g. `xmelab`.

cmarqu, 2021-02-22:
Closed by #2340.

closed 2021-02-22 by cmarqu