cocotb/cocotb #2215: Feature Request: Support for Xs

jshaker000, 2020-11-22:
Would it be possible for cocotb to define some kind of wrapper 'Logic' class that can hold a bus with X's on it, and defines and / or / etc as per the Verilog LRM? Sometimes it would be useful to be able to read back X's into python.

ktbarrett, 2020-11-22:
#2059

cmarqu, 2020-11-22:
@jshaker000 ``isunknown`` exists as https://docs.cocotb.org/en/stable/library_reference.html?highlight=x#cocotb.binary.BinaryValue.is_resolvable but it's indeed hard to find.

jshaker000, 2020-11-22:
Are nets read from the dut always 'BinaryValues' then (which can be implicitly casted to int), or how does that work?

ktbarrett, 2020-11-22:
@jshaker000 Arrays of logic's and subtypes of that (`sfixed`, `unsigned`, etc.) are `BinaryValue`. Integer nets and constants (`integer`, `natural`, etc.) return `int`. Floating point nets and constants (`real`) return `float`. Boolean nets and constants (`boolean`) return `bool`. String nets and constants (`string`) return `bytes`.

jshaker000, 2020-11-23:
Thanks, yea it could be clearer on thw wiki but this is enough to get me started. Thanks!

ktbarrett, 2020-11-23:
@jshaker `int` is a Python builtin, so that's not possible. However, `is_resolvable` could be a free function that would accept `int` or `BinaryValue`, etc.

closed 2021-01-16 by jshaker000