cocotb/cocotb #2216: Warn of upcoming deprecations

cmarqu, 2020-11-22:
We should have a section like that in the docs of the 1.5 release: https://docs.antora.org/antora/2.3/whats-new/#caution-coming-soon

_Originally posted by @cmarqu in https://github.com/cocotb/cocotb/issues/1642#issuecomment-731851364_

ktbarrett, 2020-11-24:
Could we perhaps just link to the 2.0 milestone and ensure we are marking all our great ideas for 2.0?

cmarqu, 2020-12-07:
Yes, that, and we should also [link to the deprecations](https://github.com/cocotb/cocotb/issues?q=is%3Aopen+is%3Aissue+milestone%3A2.0+label%3Atype%3Adeprecation) specifically.

cmarqu, 2021-01-18:
Warn of Python 3.6+ being required.

ktbarrett, 2021-01-25:
I don't think we need to do this since we decided to do a 1.6 release. The 'latest' documentation after release could have a warning on the home page about it being a breaking release, and of course 1.6 will be just be a bunch of warnings about what we have changed.

imphil, 2024-07-08:
* We're done with 1.9.
* I don't see much actionable work here anyways: we do have a deprecations section in the release notes, and users get notified via in-product messages. That's probably good enough, since nobody implemented anything better since this issue was created 4 years ago.

closed 2024-07-08 by imphil