cocotb/cocotb #2220: Add gcovr to tox config

cmarqu, 2020-11-23:
We could add gcovr back to show the coverage results when running tox locally.

I also think Python/C++ coverage should be optional when running locally through a `-coverage` suffix to the tox env, but that can be a different PR.

_Originally posted by @garmin-mjames in https://github.com/cocotb/cocotb/issues/2155#issuecomment-729886895_

garmin-mjames, 2020-11-23:
I have a branch printing coverage results, and making coverage non-default for local `tox` runs, but I ran into an issue with `tox-gh-actions` https://github.com/ymyzk/tox-gh-actions/issues/44.

I didn't want the `*-coverage` builds to be part of the default run, but `tox-gh-actions` only runs environments defined in `envlist`.

ktbarrett, 2020-11-23:
The default run currently runs all combinations of python version and OS. The default run is already broken, is it not?

garmin-mjames, 2020-11-23:
@ktbarrett that's true. If we don't care about `tox -l` having every combination listed, I could make a PR now.

ktbarrett, 2020-11-24:
Does coverage take a considerable amount of time? Is there ever a reason to turn it off?

garmin-mjames, 2020-11-24:
@ktbarrett on my PC I saw basically no difference in runtime.

If the coverage print-outs are added, then that will be what you see after running `tox` instead of the regression results currently shown.

Another option is to just not print out the coverage results and leave that to the user. The results are saved outside of the tox run.

imphil, 2023-11-01:
Closing as we're not using tox any more. The coverage reporting is available for local users through nox.

closed 2023-11-01 by imphil