cocotb/cocotb #2222: Track current time stage and prevent improper usage in Python

ktbarrett, 2020-11-24:
`test_readwrite_in_readonly` and `test_cached_write_in_readonly` exist to test invalid actions, like waiting for `ReadWrite` while in `ReadOnly`, and doing writes in `ReadOnly`. The issue is that doing these actions cause certain simulators to error or fail hard, which is why it is skipped in several simulators. Instead of letting the simulator decide what to do, we should keep track of the current time state and raise errors early.

garmin-mjames, 2020-11-24:
This is already done for writes in `ReadOnly`:

https://github.com/cocotb/cocotb/blob/79d6c78429628f47cb62010fd073dfd6626cd1cf/cocotb/scheduler.py#L512-L513

`_MODE_WRITE` is never used though.

ktbarrett, 2020-11-24:
Correct, the way the test was written implied that maybe the test wasn't updated after that protection was added. Additionally, that, `ReadWrite` in `ReadOnly` and `Timer(0)` all fall into the same "undefined behavior" category. We are throwing a basic `Exception` there, but using `RuntimeWarning` for `Timer(0)`. Maybe we need to make that more consistent? And also catch `ReadWrite` in `ReadOnly` to begin with.

PR #4208 (2024-10-09): Prevent transitions from ReadOnly to ReadWrite or ReadOnly
Transitioning from ReadOnly to ReadWrite or ReadOnly are not legal, this checks prevents that from occuring by raising a `RuntimeError` when it is attempted. Closes #1245. Closes #2222.
closed 2024-10-09 by ktbarrett