cocotb/cocotb #2232: Remove ping_tun_tap example and documentation?

ktbarrett, 2020-11-26:
A few reasons:

* example is out of date
  * doesn't work in Python 3 without modification
  * many newer systems don't ship with `ifconfig` tool
  * still using old generator-based coroutines
  * even with all issues resolved it still fails?
* requires sudo or other special permissions to run
  * not run in CI
* documentation is out of date
  * copy-pasted sources are no longer correct

cmarqu, 2020-11-27:
All good reasons, but it's still sad to see it go because it is/was an example showing what is easy in Python compared to other verification methods.
We should re-add such an example at some point, e.g. https://github.com/chiggs/oc_jpegencode or Matt Venn's audio example.

ktbarrett, 2020-11-27:
A similar example that doesn't require sudo permission would be nice. Either of the examples you mentioned would work. Matt's code is up to date, the JPEG encoder example is not.

cmarqu, 2020-11-27:
Matt's example has the disadvantage that it's not so ... graphic. :)

marcinwoj, 2020-11-28:
I am also a big fan of this example, as I learned from that how to intercept packets and I used something similar in my designs. 

Agree with @cmarqu this shows how powerful cocotb is to leverage the external environment and use cocotb more as a "glue" rather than a test generator.

ktbarrett, 2020-11-28:
@marcinwoj Does that mean it works for you? I would be happy if it worked, even if it did require sudo (though Windows users are hung out to dry). Could you update the test and get it back into a working state? I updated it to run under Python 3, but it didn't detect any of the incoming packets (from the `ping` subprocess) as ICMP packets and then just hung. I'm not sure if I'm missing something? I've never been able to run it successfully.

marcinwoj, 2020-11-29:
@ktbarrett yes, it worked. However, it could have been still Python 2 (not sure that now) and also I might have changed it more, but for sure I was able to pick up/send pkts from/to tun_tap devices. Vaguely remember that I used python's select.select() method somewhere there. I will dig up for the code and will try to update the example. It can take a few/several days though.

alexforencich, 2020-11-29:
Maybe we need a cocotb examples repo for stuff like this.  Instead of outright removing it, move it over there where it can be worked on independent of the core cocotb codebase.

alexforencich, 2020-12-12:
Actually, crazy idea: maybe clean this up and roll it in to https://github.com/alexforencich/cocotbext-eth.  Connecting a TAP to an AXI stream, GMII, XGMII, etc. interface could be a rather useful feature.

themperek, 2020-12-12:
> would make sense to utilize something like scapy

I have been thinking about such an example for a very long time ;-)

alexforencich, 2020-12-12:
I'm currently evaluating dpkt and scapy for use in testbenches for verilog-ethernet and corundum, but if we could also use scapy to connect to the outside world, that could certainly be an interesting capability.

eric-wieser, 2020-12-12:
There are already references to scapy elsewhere in cocotb, FWIW.

imphil, 2021-01-09:
Let's remove this example from the cocotb repositoy. From an in-tree example, I'd expect it showing a "best practice", be well described and maintained. None of that is the case for this example at the moment. Since we don't have a volunteer or other bandwidth to improve the current example, let's just remove it. We live in a world of git and eternal archives, so it's easy to resurrect it if/when the need arises.

ktbarrett, 2021-01-09:
@imphil you can review and merge #2234.

imphil, 2021-01-09:
And yes, scapy and cocotb are a good combination. A former colleague did use it e.g. here: https://github.com/aoeldemann/fluent10g

PR #2234 (2021-01-09): Remove ping TUN TAP example from repo and docs
Closes #2232. Still open for discussion in the issue whether this is a good idea or not.
closed 2021-01-09 by ktbarrett