cocotb/cocotb #2244: [Verilator] top-level signal handles differ between iteration and by-name

garmin-mjames, 2020-11-30:
I've been investigating a wrapper port vs module port issue while working on updating the main verilator loop used by cocotb (#2105).

I've been able to determine that using iteration from the module handle returns handles to the internal signals (`dff.clk`), while doing a `vpi_handle_by_name('dff.clk', NULL)` lookup will return handles to the wrapper signals (`TOP.clk`). This has to do with using the special case of an empty string name for the wrapper.

_Originally posted by @garmin-mjames in https://github.com/cocotb/cocotb/issues/2185#issuecomment-724986236_

alexforencich, 2023-01-28:
Just wondering, if this is a verilator issue and not a cocotb/GPI issue, is there a corresponding issue on the upstream verilator issue tracker?  If so, it would probably make sense to link it here.

alexforencich, 2023-01-30:
Just wondering, if iterating results in unusable handles, could you iterate, get the name, then use `vpi_handle_by_name` to get a usable handle?  Might be a decent workaround until the issue is fixed upstream in Verilator.

alexforencich, 2023-01-31:
Just tried it that idea, and it doesn't seem to work.  Basically what I did is I moved `_get_sub_handle_by_name` from HierarchyObject to RegionObject, and then I replaced most of the contents of `_discover_all` with a call to `_get_sub_handle_by_name`, which I think should mean that all of the handles that are actually used in the hierarchy would be looked up by name instead of coming from an iterator.  But, it didn't work - the result was exactly the same as before (cocotb cannot update any signals in verilator after `dir()` is called in cocotb-bus).  

But tbh, that might be a good re-factor anyway as then there is one code path for creating all handles.

marlonjames, 2023-01-31:
The GPI stores handles mapped to signal paths, and will drop any new handles that match cached paths. So even if you get the right lookup by name the second time, it most likely is throwing away the new handle because it already has the previous one that matches.

marlonjames, 2023-01-31:
And do answer your earlier question, I don't think that there is an upstream Verilator issue for this.

alexforencich, 2023-01-31:
Ah OK, so the refactoring at the python level is not helpful as it's just giving us back the garbage handle that we got previously due to the caching.  So, either something needs to be done in the GPI C++ code, or the bug needs to be reported to Verilator and fixed there.

alexforencich, 2023-02-01:
Well, I reported this to verilator as https://github.com/verilator/verilator/issues/3919 , but since I don't fully understand exactly what is going on at the VPI level, I'm going to have to defer to you guys to replicate this properly in the verilator test suite.  I cobbled together a test that seemed to have failed in the same way, but this was actually due to some verilator directives that made the reg in question explicitly read-only.

marlonjames, 2023-11-07:
This is fixed in upstream PR https://github.com/verilator/verilator/pull/4618

closed 2023-11-07 by marlonjames