cocotb/cocotb #2245: Python version update in installation notes?

raysalemi, 2020-12-01:
Python 3.5 requires the coroutine decoration, but later in the documentation it says that we no longer need the decorator, though it is allowable. 

I wonder if changing the required Python version to 3.7 would remove ambiguity for future development.

eric-wieser, 2020-12-01:
> Python 3.5 requires the coroutine decoration

Does it?

> but later in the documentation it says that we no longer need the decorator

Can you link to the bit of the docs you're looking at?

raysalemi, 2020-12-01:
The docs in 3.9 say this:

> Note Support for generator-based coroutines is deprecated and is scheduled for removal in Python 3.10.

Does this mean older testbenches won’t run with 3.10?

raysalemi, 2020-12-01:
[Here is the install](https://docs.cocotb.org/en/stable/install.html)

[And here is the later comment](https://docs.cocotb.org/en/stable/coroutines.html#generator-based-coroutines)

eric-wieser, 2020-12-01:
Can you point to the part that implies "Python 3.5 requires the coroutine decoration"?

eric-wieser, 2020-12-01:
> Does this mean older testbenches won’t run with 3.10?

Actually it doesn't - the release note you quote is specifically about `asyncio`, which cocotb does not use.

However, old cocotb is very unlikely to work against 3.10 anyway. The Python C API changes between releases, and cocotb must change to keep up. By the time cocotb supports 3.10, we might have chosen to drop support for `yield` ourselves. There's nothing stopping you running old testbenches with old python and old cocotb though.

raysalemi, 2020-12-01:
Hi Eric. I realized that I had two GitHub accounts and deleted the one in this conversation. This is the new one.

raysalemi, 2020-12-01:
I just wanted to raise the issue. If there is no need to update the required Python then we're good.

eric-wieser, 2020-12-01:
So, just to clarify things:

* You can (and should) use `async` / `await` syntax in Python 3.5 and newer. This means that if you're using the latest cocotb, which does not support earlier versions of Python, you can _always_ use this new syntax.
* The next release of cocotb is likely to be the last one to support Python 3.5, since Python 3.5 is no longer maintained. When we release after that, Python 3.5 users will probably have to upgrade to Python 3.6 (or as you suggest, jump to a newer version).
* Cocotb makes no guarantee about its releases working on versions of Python that are released after it.
* We eventually plan to rip out `yield` support entirely, when we think we can do so without breaking too many people.

Let me know if that leaves you with any questions.

raysalemi, 2020-12-01:
Yes. Though I just realized that I’m using yields because those were in the examples. I’ll go figure out how not to use them. Replace them with “await”?

cmarqu, 2020-12-01:
Yes, replace the `yield` with `await`, and the `def` with  `async def`.
The examples (and tests) of the upcoming 1.5 version of cocotb will all have been converted.

closed 2021-03-07 by eric-wieser