cocotb/cocotb #2246: Cannot run on Big Sur

raysalemi, 2020-12-01:
Cocotb cannot find libpytyon3.8.so  

I do not see it in the libs:

```
(base) raysalemi@WriteNow lib % ls libpython*
libpython3.8.a		libpython3.8.dylib	libpython3.8.nolto.a
```

Am I doing something wrong?  

* Latest version of Cocotb (just downloaded)
* MacOS 11.0.1
* icarus-verilog 11.0
* anaconda python 3.8.3
* log file:
```
MODULE=tinyalu_cocotb TESTCASE= TOPLEVEL=tinyalu TOPLEVEL_LANG=verilog \
        /usr/local/bin/vvp -M /Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp
     -.--ns ERROR    cocotb.gpi                         ..s/cocotb_utils.cpp:69   in utils_dyn_open                  Unable to open lib libpython3.8.so: dlopen(libpython3.8.so, 9): image not found

     -.--ns ERROR    cocotb.gpi                         ..mbed/gpi_embed.cpp:127  in embed_init_python               Failed to find Python shared library

     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
```

ktbarrett, 2020-12-01:
That is expected on Mac OS (#2084), but we should probably clean that up to prevent the message from printing. `libpython` is a library packaged with a Python installation, we don't package it ourselves. Are your simulations running fine?

raysalemi, 2020-12-01:
Hi Kaleb,

No. The sim isn't running, but I'm new to this and probably made some other
error.  You're saying this should not keep it from working?
Ray

ktbarrett, 2020-12-01:
@raysalemi That is correct, if something isn't working there is probably another issue occurring later. Is what you posted all that is printed?

raysalemi, 2020-12-01:
No. It hangs.

themperek, 2020-12-01:
@raysalemi 
Can try to run a simple example like https://github.com/cocotb/cocotb/tree/master/examples/simple_dff and past full log file.

themperek, 2020-12-01:
So it seems to work under CI: https://github.com/cocotb/cocotb/pull/2249/checks?check_run_id=1481370246

(The overall status if fail but this is other/setup issues).

PR #2263 (2020-12-05): Fix error on dynamic libpython loading on mac
Exiting "Error":
```
       -.--ns ERROR    cocotb.gpi                         ..s/cocotb_utils.cpp:69   in utils_dyn_open                  Unable to open lib libpython3.8.so: dlopen(libpython3.8.so, 9): image not found
       -.--ns ERROR    cocotb.gpi                         ..mbed/gpi_embed.cpp:127  in embed_init_python               Failed to find Python shared library
```
After: https://github.com/cocotb/cocotb/runs/1500502527?check_suite_focus=true#step:22:109

Closes #2246 
Related to #2247
closed 2020-12-05 by themperek