cocotb/cocotb #2248: Test designs trigger Verilator's linter

ktbarrett, 2020-12-01:
The `array_module`, `avalon_module`, and `ax4_ram` contain code that triggers Verilator's linter, which makes the tests fail, likely because of a non-0 exit code from the compiler. `test_fatal/fatal.sv` relies on a delay which Verilator doesn't support, perhaps this can be refactored to wait for a signal changed by cocotb? `plusargs_module` also uses delays, but they don't seem necessary.

marlonjames, 2021-02-10:
An easier option than all the comment pragmas is https://www.veripool.org/wiki/verilator/Manual-verilator#CONFIGURATION-FILES
These can be added to the test design folders and passed to Verilator in the Makefiles

ktbarrett, 2025-02-21:
Closed by #2264.

closed 2025-02-21 by ktbarrett