cocotb/cocotb #2251: Reading from a blocking queue in async def with fork

raysalemi, 2020-12-02:
I'm trying to create an asynchronous driver that will read transactions from a Queue.  I've got this coroutine:

```
async def bfm_send_op(dut):
    (A, B, op) = AluDriverBfm.queue.get()
    dut.A = A
    dut.B = B
    dut.op =op.value
    dut.start = 1
    print(dut.done)
    await RisingEdge(dut.done)
    await FallingEdge(dut.clk)
    dut.start = 0
```

And I launch it like this:

```
@cocotb.test()
async def test_alu(dut):
    AluDriverBfm(dut, "ALUDRIVERBFM")
    print("FORKING THINGS!")
    cocotb.fork(bfm_send_op(dut))
    cocotb.fork(Clock(dut.clk, 100).start())
    print("DONE FORKING THINGS!")
```

But I see this:

```
     0.00ns INFO     cocotb.regression                         regression.py:459  in _start_test                     Running test 1/1: test_alu
     0.00ns INFO     cocotb.test.test_alu.0x7fd592f0e4c0       decorators.py:255  in _advance                        Starting test: "test_alu"
                                                                                                                     Description: None
FORKING THINGS!
```

Investigation shows that the `fork` call is not returning because the routine blocks at the `get`.  I thought that `fork` would run my loop independently as it does with `Clock().start()`

What am I doing wrong? Why does `fork` not just leave the coroutine waiting on the `get` and move on to the next thing?

Thanks!
Ray

eric-wieser, 2020-12-02:
Are you using a `queue.Queue`? That's a threaded operation, not an async one.

`Clock().start` runs independently because it uses `await` internally. If you want simulation time to pass, you need to build an async queue, and use it as:
```
(A, B, op) = await AluDriverBfm.queue.get()
```

An async queue would be a good primitive to add to cocotb. You can probably build one on top of `collections.deque` and `cocotb.triggers.Event`.

raysalemi, 2020-12-02:
Yup. I just figured that out.  I need asyncio.Queue.

Thanks!

closed 2020-12-02 by raysalemi

eric-wieser, 2020-12-02:
`asyncio.Queue` will not work with cocotb. None of the `asyncio` stuff will.

ktbarrett, 2020-12-02:
@raysalemi A simple queue in cocotb would use a `collections.deque` and `cocotb.triggers.Event`. They are quite useful.

raysalemi, 2020-12-02:
The root issue is communicating with async methods in cocotb from a regular thread.

raysalemi, 2020-12-02:
> `asyncio.Queue` will not work with cocotb. None of the `asyncio` stuff will. `async` / `await` is a language feature. `asyncio` is _one_ implementation of how to use that language feature. `cocotb` is another. They're not compatible, sadly.

Does cocotb have the equivalent of asyncio.run()

alexforencich, 2020-12-02:
cocotb.fork?

alexforencich, 2020-12-02:
Ah, nevermind, that's something different. You want to be able to await from outside of cocotb. Not sure if there is a way to do that.

raysalemi, 2020-12-02:
It's possible at least if you launch the wait in a thread.

alexforencich, 2020-12-02:
Actually, maybe what you can do is replace the call to get with a loop around a nonblocking call to get, and wait for the next clock edge every time the queue is empty. That way, you can use a non-asyncio queue.

raysalemi, 2020-12-02:
I tried something like that, but it didn't work reliably.

alexforencich, 2020-12-02:
Didn't work reliably? What do you mean, and what did you do, exactly? I did the same thing myself (loop with await clock edge) before rewriting to await an event and it worked fine.

raysalemi, 2020-12-02:
I don't have a clock edge at that point in the flow so I used a sleep().
But it really didn't work.  I like the thread solution because there is no
questions about timing.

alexforencich, 2020-12-02:
AFAIK cocotb is single threaded, you're talking about an actual separate thread and not a coroutine, right?

raysalemi, 2020-12-02:
Yes. My stimulus is coming out of an existing thread. I want to use it to
drive pins in CocoTB.  My stimulus thread is not async and can't be async.
So I need to create a new event loop in a thread and use it to call my
coroutine driver.

alexforencich, 2020-12-02:
Okay, so why are you trying to do this before there is a clock edge?

raysalemi, 2020-12-02:
I don' thave a clock in my single thread flow. So I can't call put because
it is a coroutine.

alexforencich, 2020-12-03:
Well, how about using the asyncio queue, but call put_nowait and get_nowait on the cocotb side?

raysalemi, 2020-12-03:
It’s the put on the TLM side that breaks that. Otherwise I have to loop
over a sleep and I just don’t like that solution.

alexforencich, 2020-12-03:
Then use a normal thread-safe queue with a blocking put() and loop on the cocotb side with nonblocking get(False)?

raysalemi, 2020-12-03:
Interesting idea!

cmarqu, 2020-12-03:
@raysalemi FWIW, we also have a chat room at https://gitter.im/cocotb/Lobby where this might be easier and quicker to discuss.

alexforencich, 2020-12-03:
FWIW, gitter is kinda terrible.  Threading (heh) is not working correctly on my phone app at the moment.  And it keeps refusing to mark messages as read.