cocotb/cocotb #2252: "Got bad yield" on module output signal

raysalemi, 2020-12-02:
I don't understand this error message:

```
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/Users/raysalemi/PycharmProjects/pyuvm/tinyalu_example/tests/tinyalu_cocotb.py", line 21, in send_op
    await RisingEdge(self.dut.done)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/triggers.py", line 123, in __await__
    return (yield self)
RuntimeError: Task got bad yield: RisingEdge(ModifiableObject(tinyalu.done))
```

My DUT is named tinyalu and there is an output signal named `done`. I want to wait for the `done` signal so I do this:

```
    async def send_op(self, A, B, op):
        print("Sending: ", (A, B, op))
        self.dut.A = A
        self.dut.B = B
        self.dut.op =op.value
        self.dut.start = 1
        await RisingEdge(self.dut.done)
        await FallingEdge(self.dut.clk)
        self.dut.start = 0
        return self.dut.result
```

I get this:

```
2
clk: 0
clk: 1
3
clk: 0
reset: 1
Sending:  (213, 203, <ALUOps.ADD: 1>)
Exception in thread uvm_test_top.env.agent.dr_h:
Traceback (most recent call last):
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File ".../tinyalu_example/tests/tinyalu_tb.py", line 106, in run_it
    result = self.tpool.submit(asyncio.run, self.bfm.send_op(command.A, command.B, command.op)).result()
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/concurrent/futures/_base.py", line 439, in result
    return self.__get_result()
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/concurrent/futures/_base.py", line 388, in __get_result
    raise self._exception
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File ".../tinyalu_example/tests/tinyalu_cocotb.py", line 21, in send_op
    await RisingEdge(self.dut.done)
  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/triggers.py", line 123, in __await__
    return (yield self)
RuntimeError: Task got bad yield: RisingEdge(ModifiableObject(tinyalu.done))
```

Any idea why?

eric-wieser, 2020-12-02:
> ```
>   File "/Users/raysalemi/opt/anaconda3/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
>     return future.result()
> ```

You are mixing `asyncio` with `cocotb`. This is not supported.

ktbarrett, 2020-12-02:
@raysalemi I see `asyncio` in your traceback. cocotb's scheduler and triggers are not compatible with anything from `asyncio`. `asyncio`'s scheduler isn't sufficient for what cocotb needs to do. If you need a queue you'll have to implement a simple one yourself https://github.com/cocotb/cocotb/issues/2251#issuecomment-737217582, until an implementation is added here. This is a FOSS project, dependent upon volunteers; feel free to open a pull request if you get something working and we may incorporate it into the repo.

raysalemi, 2020-12-02:
Aha! 

I used the asyncio.run method. What would be needed in a new run method to work with cocotb? Can you point me in the right direction?

eric-wieser, 2020-12-02:
Cocotb does not really have a `run` method right now. To run cocotb code, you put it inside a function decorated with `@cocotb.test()`

raysalemi, 2020-12-02:
I need to call an async routine from a regular thread. So I do this:

```
            result = self.tpool.submit(asyncio.run, self.bfm.send_op(command.A, command.B, command.op)).result()
```

This creates a new thread with a new event loop. So my loopless thread can get data from an async thread.

Does cocotb have a mechanism for implementing this?  That is the real ask here, and the thing I'd be happy to write with some guidance.

ktbarrett, 2020-12-02:
Using threads is discouraged. cocotb has [`cocotb.fork`](https://docs.cocotb.org/en/stable/coroutines.html) if you need to run a coroutine concurrently. A forked coroutine executes independently and simultaneously to other coroutines, in terms of sim time. Unlike threads, they are safe from most data races since they aren't _actually_ running simultaneously. Forked coroutines are coroutines (`async def`) and you can `await` in them.

eric-wieser, 2020-12-02:
If you want threads, you need `cocotb.external` and `cocotb.function`.

ktbarrett, 2020-12-02:
@raysalemi Then you will need [`cocotb.external` and `cocotb.function`](https://docs.cocotb.org/en/stable/library_reference.html?highlight=cocotb.external#cocotb.coroutine). `cocotb.external` turns a regular function into a coroutine so you can `await` it, and `cocotb.function` allows you to call a coroutine like a regular function. In your case you would wrap a coroutine you intend to call from your thread pool (assuming that's what `tpool` is) with `cocotb.function`.

raysalemi, 2020-12-02:
Aha!  Thanks. I'll check it out.

raysalemi, 2020-12-02:
> @raysalemi Then you will need [`cocotb.external` and `cocotb.function`](https://docs.cocotb.org/en/stable/library_reference.html?highlight=cocotb.external#cocotb.coroutine). `cocotb.external` turns [...]

I'm afraid this didn't solve the problem.  This works great if I want to call a function that calls a coroutine.  But I want to call a function that launches a bunch of threads and one of those threads calls the coroutine. Down in that thread I have a regular function that needs to call await to launch the external function and now I'm stuck needing a coroutine again.

I'll take a look at asyncio.run() and see if I can emulate it.  Or made launch the external function in a thread.

eric-wieser, 2020-12-02:
> But I want to call a function that launches a bunch of threads and one of those threads calls the coroutine.

So, what I'd do here is:
* Launch that function in a new thread (X) with `external`
* Spawn off the child threads as before
* Have the child thread communicate to thread X that it has coroutines to run

What are you actually trying to do here? It doesn't sound like your concurrency model makes sense in the context of simulator time steps.

raysalemi, 2020-12-02:
The model is intended to be a pure TLM testbench with blocking threads.

raysalemi, 2020-12-02:
> So, what I'd do here is:
>
> * Launch that function in a new thread (X) with `external`
> * Spawn off the child threads as before
> * Have the child thread communicate to thread X that it has coroutines to [...]

What I'd really like is to make a non-blocking call to a function that launches a blocking coroutine.  Then my thread blocks until notified that the coroutine has finished its work.  The mistake I made here was putting the RisingEdge into the function I called.  

I think I need to figure out how to do a trigger without needing an async function to do it.

raysalemi, 2020-12-02:
Another thing that would work is to write a coroutine that loops on the clock and waits to see a variable that says, "You have a transaction, go handle it."

raysalemi, 2020-12-03:
Hi All. 

Thank you for your help.  I now have an example that illustrates both what I'm trying to do and the problems I'm seeing. I've implemented @alexforencich's idea to use a blocking `put` on the thread and a `get_nowait` on the coroutine side.

I've got a DUT that simply takes in numbers and writes them out:
```
module dut(input bit clk,
input bit reset_n, input int unsigned data_in, output int unsigned data_out);
    always @(posedge clk)
        if (reset_n == 0)
            begin
                $display("reseting");
                data_out <= 0;
            end
        else
            begin
                $display("data_in: %0d", data_in);
                data_out <= data_in;
            end

endmodule : dut
```

And I test it with this cocotb testbench:

```
otb
import test_sw
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge, FallingEdge
from queue import Queue, Empty
import tester

class CocoTBBFM():

    def __init__(self, dut):
        self.dut = dut
        self.queue = Queue(maxsize=1)

    def send_num(self, num):
        self.queue.put(num)

    async def start(self):
        while True:
            await RisingEdge(self.dut.clk)
            try:
                numb = self.queue.get_nowait()
                self.dut.data_in = numb
            except Empty:
                pass

    async def reset(self):
        await FallingEdge(self.dut.clk)
        self.dut.reset_n = 0
        await FallingEdge(self.dut.clk)
        self.dut.reset_n = 1
        await RisingEdge(self.dut.clk)

@cocotb.external
def run_test(test):
    tester.Tester.run_test(test)

@cocotb.test()
async def test_alu(dut):
    clock = Clock(dut.clk, 2, units="us")
    cocotb.fork(clock.start())
    bfm = CocoTBBFM(dut)
    stim = test_sw.Stim(5,dut)
    await bfm.reset()
    cocotb.fork(bfm.start())
    await run_test(stim.numb_gen_test)
```

The stimulus generator is launched as a thread from the tester:

```
import threading

class Tester():
    @staticmethod
    def run_test(test):
        tt = threading.Thread(target=test, name="test_thread")
        tt.start()
        tt.join()

```

The result either hangs as soon as I call run_test. If I use cocotb.function on send_test() I get this:

  File "/Users/raysalemi/opt/anaconda3/lib/python3.8/site-packages/cocotb/decorators.py", line 102, in __init__
    raise TypeError(
TypeError: None isn't a valid coroutine! Did you forget to use the yield keyword?

I cannot find a combination of tools that make this work without hanging.

[ray_example.tar.gz](https://github.com/cocotb/cocotb/files/5636571/ray_example.tar.gz)

eric-wieser, 2020-12-03:
I uploaded you code into a gist at https://gist.github.com/eric-wieser/3664893b77b43a3ff89ecd1653478017 to make it easier to look at

eric-wieser, 2020-12-03:
It doesn't make sense to me why you want to do `threading.Thread(target=test, name="test_thread")` - running a test anywhere other than the simulator thread doesn't make sense, as you only have one simulator running, so only can run one test at a time.

raysalemi, 2020-12-03:
The application I have uses a variety of threads to generate stimulus.

eric-wieser, 2020-12-03:
1. Do those threads need to touch the HDL, and if so, why?
2. Should the simulation pause until the stimulus is ready? If not, how many simulator clock cycles per thread-time second should it wait?

raysalemi, 2020-12-03:
No. The threads don't directly access the hdl.  They create transactions that get sent to the BFM. Then the thread blocks until the BFM says that it's ready for the next transaction.

Normally this is done with DPI.  You call a function that puts the data on the BFM and the BFM checks it on each clock edge.  Then the BFM does a function call to notify the thread that it can continue.  

So the simulation runs and polls in the HDL and the TLM side makes function calls and then blocks.

It seems there is no way to get cocotb to launch the stim thread and release control and have another part of the system polling for data.

garmin-mjames, 2020-12-03:
A few things I noticed:
1. `stim.bfm` is a different object than `bfm` in `test_alu()`, so you're putting data in one queue and trying to get data from a different queue.
2. `@cocotb.external` already runs the function in a separate thread. Your code then creates another new thread and tries to call a `@cocotb.function`-decorated function from that new thread, which won't work. The cocotb scheduler expects those to be called from the thread started with `external`.
3. `Stim.num_gen_test` should be a coroutine. `@cocotb.function` wraps coroutines, which are essentially forked in the main cocotb thread when called from another thread that was started with `@cocotb.external` (and blocks that thread until the forked coroutine finishes).
4. With those fixed, `send_num` still blocks the the coroutine from completing because the queue is max size of 1, and so `start` is not able to run and complete the transaction.

raysalemi, 2020-12-03:
Can the thread send data straight through the cocotb dut without needing to
call a coroutine?

Maybe all the scheduling should happen in the HDL.

garmin-mjames, 2020-12-03:
I wouldn't access the `dut` outside the main thread, the GPI isn't thread-safe.
If you need data from other threads, I would do what @eric-wieser suggested and use a thread-safe queue and convert `start` into an `external` function, spawning `function` coroutines for each transaction.

raysalemi, 2020-12-03:
Hi Marlon,

What do you mean when you say "spawning function coroutines for each
transaction?"

Ray

raysalemi, 2020-12-03:
I found my bug. It had nothing to do with COsimulation.  It was because I am a dope:

This created a Queue:
```
class Stim():
    def __init__(self, max, dut):
        self.max = max
        self.bfm = dut_cocotb.CocoTBBFM(dut)

```

And this created a different Queue:

```
async def test_alu(dut):
    clock = Clock(dut.clk, 2, units="us")
    cocotb.fork(clock.start())
    bfm = CocoTBBFM(dut)
```

So I was writing to one queue but not reading from the same queue. It was a legitimate hang.  

@mballance, FYI

raysalemi, 2020-12-03:
Here is the secret sauce:

```
def run_test(test):
    tester.Tester.run_test(test)

@cocotb.test()
async def test_alu(dut):
    clock = Clock(dut.clk, 2, units="us")
    cocotb.fork(clock.start())
    bfm = CocoTBBFM(dut)
    stim = test_sw.Stim(5,dut, bfm)
    await bfm.reset()
    cocotb.fork(bfm.start())
    tt = threading.Thread(target=run_test, args=(stim.numb_gen_test,), name="Run Thread")
    tt.start()
    await bfm.done.wait()
```

So I made a thread and then used an event to keep the test alive as long as the thread wanted it alive.

```
    def numb_gen_test(self):
        for ii in range(self.max):
            self.bfm.send_num(ii)
        time.sleep(5)
        self.bfm.done.set()
```

raysalemi, 2020-12-04:
**Thank you all!**

My testbench is running perfectly.  

1. I confused myself by thinking that there were coroutine problems when it was my own bugs.
2. The key was to put an event at the end of the cocotb test and then trigger it from my stimulus when I was done.

Thanks agains!

closed 2020-12-04 by raysalemi