cocotb/cocotb #2259: Separate documentation on makefile-based automation and cocotb configuration

ktbarrett, 2020-12-04:
Currently, the documentation for the makefile-based project automation and the environment variables which configure the cocotb runtime are co-located. To some degree this makes sense, the environment variables are usually set in a user's makefiles and exported. However, to better show the separation between cocotb and the makefiles for documenting custom flows, making these two topics into two separate pages will help understanding.

ktbarrett, 2020-12-04:
RFC @cmarqu.

cmarqu, 2020-12-04:
Splitting these would be a good change.
We should take a bit care to not overcrowd the Sphinx sidebar, but if we are getting tight there, we could e.g. collect the "Python Code Library Reference" and "GPI Library Reference" pages under a single heading.