cocotb/cocotb #2262: Riviera failing because of #2256 ?

themperek, 2020-12-04:


garmin-mjames, 2020-12-04:
Looks like this is the culprit for the Verilog run:
```
# ALOG: Error: VCP2979 tests/designs/sample_module/sample_module.sv : (140, 1): Variable 'inout_if' has more than one continuous driver.
```

garmin-mjames, 2020-12-04:
This is also affecting Verilator CI:
https://github.com/cocotb/cocotb/pull/2105/checks?check_run_id=1500223475#step:22:112

garmin-mjames, 2020-12-04:
maybe the port needs to be `inout` instead `input`?

cmarqu, 2020-12-04:
> maybe the port needs to be `inout` instead of `input`?

That could be a solution (and then the direction would match the port name), but in #2265 at least I just wanted to do the minimal change necessary (behavior-wise, not code-wise).

PR #2265 (2020-12-05): Remove erroneous assignment to input port
Interestingly, both Xcelium and Questa allow that
while Verilator and Riviera show an error.

Closes #2262.
commit c89e1de0 (2020-12-05): Remove erroneous assignment to input port (#2265)

Interestingly, both Xcelium and Questa allow that
while Verilator and Riviera show an error.

Closes #2262.
closed 2020-12-05 by cmarqu