cocotb/cocotb #2267: How to get the example ping_tun_tap work? and the matrix_multiplier?

ganghuang, 2020-12-05:
**### The testing system**
* the cocotb version used,
Just a new git clone from the repo, git log: d97e9414cf7721e3e1b0ef272cbda21f06521d62

* the operating system and version (32/64 bit),
Ubuntu 20.04.1 
and 
Debian 10

* the simulator and version (32/64 bit),
iverilog 10.2
Verilator 4.101 devel rev v4.100-29-gc5da072f

* the Python version, and where it's coming from (e.g. system, Anaconda, self-installed, ...),
python 3.8.5
python 3.7.3

* (part of) the log file (cleaned of proprietary information).
**### What is working**
I am just start playing on the cocotb, but I found I can not run all the examples.
In the example directory, 
adder
axi_lite_slave
dff
simple_dff
endian_swapper
mean  
mixed_language
all seems working fine, as it doesn't report error and reach a shutting down

matrix_multiplier   ping_tun_tap  
### **ping_tun_tap**
For the ping_tun_tap, I can not get it work. 
It first report 
File "/tmp/cocotb/cocotb/examples/ping_tun_tap/tests/test_icmp_reply.py", line 48, in create_tun
                                                                                                                         tun = open('/dev/net/tun', 'r+b')
                                                                                                                     io.UnsupportedOperation: File or stream is not seekable.

I searched and found adding buffering=0 seems can solve this.
Then it report 

File "/tmp/cocotb/cocotb/examples/ping_tun_tap/tests/test_icmp_reply.py", line 56, in create_tun
                                                                                                                         ifr = struct.pack('16sH', name, IFF_TUN | IFF_NO_PI)
                                                                                                                     struct.error: argument for 's' must be a bytes object

I add a encode('utf-8') there
So the git diff is like this now:

diff --git a/examples/ping_tun_tap/tests/test_icmp_reply.py b/examples/ping_tun_tap/tests/test_icmp_reply.py
index 8f516a7b..f2b0a25d 100644
--- a/examples/ping_tun_tap/tests/test_icmp_reply.py
+++ b/examples/ping_tun_tap/tests/test_icmp_reply.py
@@ -45,7 +45,7 @@ def create_tun(name="tun0", ip="192.168.255.1"):
     TUNSETOWNER = TUNSETIFF + 2
     IFF_TUN = 0x0001
     IFF_NO_PI = 0x1000
-    tun = open('/dev/net/tun', 'r+b')
+    tun = open('/dev/net/tun', 'r+b', buffering=0)
     tun_num = int(name.split('tun')[-1])
 
     # Try and create tun device until we find a name not in use
@@ -53,7 +53,9 @@ def create_tun(name="tun0", ip="192.168.255.1"):
     while True:
         try:
             name = 'tun{}'.format(tun_num)
-            ifr = struct.pack('16sH', name, IFF_TUN | IFF_NO_PI)
+            print(type(name))
+            print(name)
+            ifr = struct.pack('16sH', name.encode('utf-8'), IFF_TUN | IFF_NO_PI)
             cocotb.log.info(name)
[... 84 lines cut ...]
VCD warning: array word matrix_multiplier.c_o[19] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[20] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[21] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[22] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[23] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[24] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[25] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[26] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[27] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[28] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[29] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[30] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[31] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[32] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[33] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[34] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[35] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[36] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[37] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[38] will conflict with an escaped identifier.
VCD warning: array word matrix_multiplier.c_o[39] will conflict with an escaped identifier.
ERROR: results.xml was not written by the simulation!
make[1]: *** [/usr/local/lib/python3.8/dist-packages/cocotb/share/makefiles/simulators/Makefile.icarus:69: results.xml] Error 1
make[1]: Leaving directory '/tmp/cocotb/cocotb/examples/matrix_multiplier/tests'
make: *** [/usr/local/lib/python3.8/dist-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2

ktbarrett, 2020-12-05:
The `ping_tun_tap` example is terribly out-of-date and I've been unable to get it working again. We are considering removing it #2234.

> iverilog 10.2

There is still some missing functionality in that old of a version of iverilog that I cannot be certain everything works. Try a newer version of Icarus, like 10.3 or 11.0.

> Verilator 4.101

All versions of Verilator before 4.106 are unsupported and don't really work. Verilator support should also improve dramatically once #2105 is merged.

cmarqu, 2020-12-05:
The `ping_tun_tap` example is known to be broken, but maybe there is a chance to revive it, see https://github.com/cocotb/cocotb/issues/2232.

For the second example, you are using the example code from cocotb master (or at least a version newer than 1.4.0), but the cocotb you are running is 1.4.0 - see the line `0.00ns INFO cocotb init.py:208 in _initialise_testbench Running tests with cocotb v1.4.0 from /usr/local/lib/python3.8/dist-packages/cocotb` in your output.

Ah, and what @ktbarrett said :)

ganghuang, 2020-12-08:
Thank you for the help. And I was thinking about use this for my code development.
So I upgrade the verilator to the latest Verilator 4.107 devel rev v4.106-22-gcd248f6b
and start to write a tap simulation based on the example.

The attached is my start don't really have the verilog side attached. I can see the tap is setup ok, and the frames are coming from the tap the .sv. But I got also got multiple basic problems here, maybe just because I am new to the cocotb:
1. The clock occasionally jump as missing one cycle. I think I am using the await wrong, can you help me on that?
2. Even the simulation only last ~4us, but the .vcd has zeros all the way to 400us, I don't know why so.
3. For the TAP, I was expecting the preamble 0x55 and SFD 0xd5 there, but if I don't have the IFF_NO_PI, I got the 0x00008000 at the beginning. So I have to manually add the 0x555555d5 there. 
[image]

[tapsim.tar.gz](https://github.com/cocotb/cocotb/files/5656020/tapsim.tar.gz)

closed 2021-03-14 by github-actions[bot]