cocotb/cocotb #2271: Supporting PyPy

ktbarrett, 2020-12-08:
To perhaps improve performance slightly within Python we could provide the ability to run with PyPy over the CPython interpreter. PyPy's [embedding process](https://doc.pypy.org/en/improve-docs/embedding.html) will require us to radically simplify our C API code. But we can expect [~4.2x increase in performance](https://www.pypy.org/) from the Python part of the testbench.

ktbarrett, 2021-02-18:
This will also require us to replace the simulator module with a CFFI or ctypes interface. This probably best belongs in an extension once we have add the capability to the GPI to load into something other than the embed library (#1994).

marlonjames, 2021-04-07:
When mature enough, HPy would allow an extension that supports both PyPy and CPython.
https://hpyproject.org/

ktbarrett, 2021-04-07:
However you would lose the speed benefits of using an FFI. `cffi` is also usable with both PyPy and CPython. If we can simplify the embedding process in `gpi_embed` we can easily support both PyPy and CPython with an `ifdef`.