cocotb/cocotb #2276: COMPILE_ARGS: VHDL and Verilog co simulation on Modelsim

Botnic, 2020-12-08:
Hello 

I run Cocotb with Modelsim. My code is VHDL 2008, so I need to pass the argument -2008 to vsim.

```
SIM = modelsim
ARCH = i686
TOPLEVEL_LANG = vhdl
COMPILE_ARGS += -2008
SIM_ARGS += -t ps
```
The Makefile configuration above generates the runsim.do with the following code snippet: 
`vcom -work work -2008 +acc <all my VHDL files>`
So far so good.

Now I integrated some code from a colleague written in Verilog.
Problem: 
The compilation command for the verilog files looks as follows:
```
vcom -work work -2008 +acc <all my VHDL files>
vlog -work work +define+COCOTB_SIM -sv -timescale 1ns/1ps -mfcu -2008 +acc <all my verilog files>
```

vlog now complains that it doesn't know the argument -2008. If I delete it by hand, the compilation and simulation works fine.

Is there a possibility to pass a COMPILE_ARGS depending on the language?
Thank you

Aditional system information: 
- CocoTB Version 1.5.0 dev0 
- OS: 64 bit 
- Simulator: Modelsim 32bit DE 2020.4
- Python: 3.7.7

themperek, 2020-12-08:
>Is there a possibility to pass a COMPILE_ARGS depending on the language?

Unfortunately not.

What should be able to do is copy https://github.com/cocotb/cocotb/blob/master/cocotb/share/makefiles/simulators/Makefile.questa modify as you like and include at the end of `Makefile` instead of `Makefile.sim`.

Botnic, 2020-12-08:
Hello Themperek
Thank you for your quick answer. 
You brought me to the right path: 

I removed the `COMPILE_ARGS += -2008` part and replaced it by `VCOM_ARGS += -2008`.
Now VLOG_ARGS is not touched and the simulation works :-)

closed 2020-12-08 by Botnic

ktbarrett, 2022-07-07:
@alexgomp Just to reiterate @themperek's point, the Makefiles have nothing at all to do with how cocotb functions. Please feel free to use whatever you feel like: modified Makefiles, custom shell scripts, the new Python runner, whatever... I most certainly do not use the Makefiles at work, neither does @marlonjames. They kinda suck and are really only there for cocotb's internal testing and to give new users *something* to get them up and running.

alexgomp, 2022-07-11:
@themperek and @ktbarrett thank you for your quick answers.

That's nice to hear that Makefiles are not actually intended to be required for using CocoTB. I agree, they can be pretty cumbersome.

If I understand you correctly then, Makefiles and their associated variables as described in the documentation, https://docs.cocotb.org/en/stable/building.html#makefile-based-test-scripts **_are not_** the intended user entry-point / interface for setting up a simulation. 

What is then the correct entry-point? Is there still work open to fully document the entry-point?

ktbarrett, 2022-07-11:
@alexgomp We have some documentation [here](https://docs.cocotb.org/en/latest/custom_flows.html). The basics are...
1. Install cocotb with pip.
2. Call your simulator passing a flag to load the GPI implementation you need use whatever pli flag your simulator uses.
3. You may need to set `LIBPYTHON_LOC` if you get an exception when starting up that mentions a failure to `import site`.
4. Required environment variables to get cocotb tests to run are `TOPLEVEL` and `MODULE`.

In the [variables documentation](https://docs.cocotb.org/en/latest/building.html), the variables are used by cocotb, but the "Makefile-based Test Scripts" are for the Makefiles only and can be ignored. We really should split the two sections into different pages to make that more obvious.