cocotb/cocotb #2277: FLI library problem when running cocotb on windows 10

htakeugm, 2020-12-08:
Hello,
1) My environment setup is as follow:
* the cocotb version used is 1.4.0
* the operating system is windows 10 64 bit
* the simulator is questasim 10.7a win64
* the Python version: Python 3.7.4 (default, Aug  9 2019, 18:34:13) [MSC v.1915 64 bit (AMD64)] :: Anaconda, Inc. on win32
* My anaconda version is: conda 4.9.2

2) I install cocotb by following the link: https://docs.cocotb.org/en/stable/install.html
 by typing the following two command from anaconda Prompt:
 - conda install -c msys2 m2-base m2-make m2w64-toolchain libpython
 - pip install --global-option build_ext --global-option --compiler=mingw32 cocotb

3) Then i download cocotb examples from: git clone https://github.com/potentialventures/cocotb.git

4) I choose to run the "adder" example by typing the following command from anaconda Prompt:
make SIM=questa GUI=1 TOPLEVEL_LANG=vhdl

5) And this is the output i received:
make results.xml
make[1]: Entering directory '/c/Users/201002810/Downloads/cocotb-master1/cocotb-master/examples/adder/tests'                         **ERROR**: cocotb was not installed with an FLI library, as the mti.h header could not be located.
If you installed an FLI-capable simulator after cocotb, you will need to reinstall cocotb.
Please check the cocotb documentation on ModelSim support.
/c/ProgramData/Anaconda3/lib/site-packages/cocotb/share/makefiles/simulators/Makefile.questa:75: recipe for target '/c/ProgramData/Anaconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll' failed
make[1]: *** [/c/ProgramData/Anaconda3/lib/site-packages/cocotb/libs/libcocotbfli_modelsim.dll] Error 1
make[1]: Leaving directory '/c/Users/201002810/Downloads/cocotb-master1/cocotb-master/examples/adder/tests'
C:\ProgramData\Anaconda3\lib\site-packages\cocotb\share\makefiles/Makefile.inc:39: recipe for target 'sim' failed
make: *** [sim] Error 2

6) Question: 
6-a) Why coctb doesn't found the FLI library? The test pass for verilog and not for VHDL
6-b) I went to directory Anaconda3/lib/site-packages/cocotb/libs and i didn't see the DLL "libcocotbfli_modelsim.dll". Why?
Can you send me the FLI DLL and tell me in which directory i should save it?

Thanks for you answer

ktbarrett, 2020-12-08:
cocotb ***must*** be installed with Questa in your path for it to compile the FLI interface. If you have Questa on your path issue the command `pip install --force-reinstall cocotb`.

closed 2020-12-08 by ktbarrett

htakeugm, 2020-12-08:
Thanks,
The test pass now for VHDL source code.
questa (C:\questasim64_10.7a\win64) was already been set on my path.
 And now i can see the FLI library into site-packages/cocotb/libs after running the command you send to me

[image]

[image]

ktbarrett, 2020-12-08:
We did add more information to the error message in the next version, so hopefully it will be less confusing soon :)