cocotb/cocotb #2278: Segregate public and private interfaces

ktbarrett, 2020-12-09:
In Python, the use of methods and attributes starting with a single underscore (`_method()`) are used to communicate that method or attribute is a private implementation detail and should not be used by the user. Sphinx, Python linters, Python REPLs, and more respect this convention. However, the code in this repo does not. The use of non-underscore methods and attributes on what are clearly private interfaces is rampant. During the 2.0 development phase this should be cleaned up.

alexforencich, 2020-12-09:
And the other way around as well - for example `cocotb.utils._get_simulator_precision()` maybe shouldn't start with an underscore.

ktbarrett, 2025-02-18:
I am going to mark this as done as a lot of work has been done on this and public APIs are good. It may not be perfect, but we will leave that to individual issues.

closed 2025-02-18 by ktbarrett