cocotb/cocotb #2279: Failing tests give a return code of zero

toddstrader, 2020-12-09:
I broke `simple_dff` like so:

```
$ git diff
diff --git a/examples/simple_dff/dff.sv b/examples/simple_dff/dff.sv
index 442389f7..559f8a2e 100644
--- a/examples/simple_dff/dff.sv
+++ b/examples/simple_dff/dff.sv
@@ -9,7 +9,7 @@ module dff (
 );
 
 always @(posedge clk) begin
-  q <= d;
+  q <= ~d;
 end
 
 endmodule
```

Naturally, the test fails, but the process yields a return code of `0`:
```
$ make SIM=verilator
make results.xml
make[1]: Entering directory '/path/to/cocotb/examples/simple_dff'
MODULE=test_dff TESTCASE= TOPLEVEL=dff TOPLEVEL_LANG=verilog \
        sim_build/dff 
. . .
 15000.00ns ERROR    cocotb.regression                         regression.py:399  in _score_test                     Test Failed: test_dff_simple (result was AssertionError)
                                                                                                                     Traceback (most recent call last):
                                                                                                                       File "/path/to/cocotb/examples/simple_dff/test_dff.py", line 22, in test_dff_simple
                                                                                                                         assert dut.q == val, "output q was incorrect on the {}th cycle".format(i)
                                                                                                                     AssertionError: output q was incorrect on the 0th cycle
 15000.00ns ERROR    cocotb.regression                         regression.py:479  in _log_test_summary               Failed 1 out of 1 tests (0 skipped)
 15000.00ns INFO     cocotb.regression                         regression.py:551  in _log_test_summary               **********************************************************************************
                                                                                                                     ** TEST                      PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                                                                                                                     **********************************************************************************
                                                                                                                     ** test_dff.test_dff_simple    FAIL        15000.00          0.00   23037189.38  **
                                                                                                                     **********************************************************************************
                                                                                                                     
 15000.00ns INFO     cocotb.regression                         regression.py:568  in _log_sim_summary                *************************************************************************************
                                                                                                                     **                                 ERRORS : 1                                      **
                                                                                                                     *************************************************************************************
                                                                                                                     **                               SIM TIME : 15000.00 NS                            **
                                                                                                                     **                              REAL TIME : 0.00 S                                 **
                                                                                                                     **                        SIM / REAL TIME : 10813778.65 NS/S                       **
                                                                                                                     *************************************************************************************
                                                                                                                     
 15000.00ns INFO     cocotb.regression                         regression.py:259  in tear_down                       Shutting down...
- :0: Verilog $finish
make[1]: Leaving directory '/path/to/cocotb/examples/simple_dff'
$ echo $?
0
```

The same thing happens for Icarus.

My primary interest is with Verilator, but it would be good if the problem could be solved in a generic way.  If I were only concerned about Verilator, I would just expose a function from a cocotb library so that the Verilator `main()` could ask cocotb if there were failures.  I'm not sure if there is a VPI way to tell the simulator to terminate with a non-zero return code.  The closest I see is `vpi_control()` but that just allows you to call `$stop`, `$finish` and `$reset` which doesn't seem like it will help.

Questions:

- Is there a simulator agnostic way to do this?
- If not, is a Verilator-only solution acceptable?

eric-wieser, 2020-12-09:
We solve this problem in CI by looking at the resulting XML file that contains the test information. I don't think VPI has anything to offer us here.

toddstrader, 2020-12-10:
Gotcha.  One other note for completeness: my current workaround to this (didn't mention before because it's kind of ugly) is to `kill(getpid(), SIGTERM)` from `VpiImpl::sim_end()` if failures are detected.  I'll either maintain this locally or add some XML-checking wrapper scripts to our flow.  But if the brute-force solution would be of interest as an optional mode I'd be happy to work up a PR.

closed 2020-12-10 by toddstrader

ktbarrett, 2020-12-10:
@toddstrader You should maintain that locally. For a fact I know that will break Questa badly.