cocotb/cocotb #2280: [Verilator] vpiReg returned for real and string types

garmin-mjames, 2020-12-10:
Verilator is returning `vpiReg` from `vpi_get(vpiType, hdl)` for `real` and `string` variables.
cocotb then creates a `ModifiableObject` instead of a `RealObject` or `StringObject`.
Assignment with a `float` or `bytes` value then raises a `TypeError`.

ktbarrett, 2020-12-10:
I don't think that is *wrong* per se, but the basic VPI without the SV extensions has no way to map those regs to specific types. The SV extensions support `vpiStringVar`, `vpiRealNet`, `vpiIntegerNet` and so on (from [here](https://github.com/cocotb/cocotb/blob/f8c0e402dd4e4f0193be81e9e4c37e0e2402f95e/cocotb/share/lib/vpi/VpiImpl.cpp#L98). This is work that needs to be done in Verilator. It's an SV simulator, it should work with the SV VPI extensions.

marlonjames, 2023-10-24:
Upstream PR to add `vpiRealVar` and `vpiStringVar` support: https://github.com/verilator/verilator/pull/4594

marlonjames, 2023-10-25:
Upstream PR has been merged.

closed 2023-10-25 by marlonjames