cocotb/cocotb #2282: [Verilator] array ranges always high index to low index

garmin-mjames, 2020-12-11:
Simulator: Verilator 4.106 64-bit

Verilator is returning the high index for `vpiLeftRange` and the low index for `vpiRightRange` regardless of the order of the range specification.

Given the following unpacked arrays:
```systemverilog
reg [7:0]  array_7_downto_4[7:4];
reg [7:0]  array_4_to_7[4:7];
reg [7:0]  array_3_downto_0[3:0];
reg [7:0]  array_0_to_3[0:3];
```

Printing the `_range` property of the handles, which consists of a `(vpiLeftRange, vpiRightRange)` tuple:
```
dut.array_7_downto_4 range is (7, 4)
dut.array_4_to_7     range is (7, 4)
dut.array_3_downto_0 range is (3, 0)
dut.array_0_to_3     range is (3, 0)
```

ktbarrett, 2020-12-11:
Doesn't verilator warn about ascending indexes? It seems to have some opinions as to what is good or not.

garmin-mjames, 2020-12-11:
It warns about 1-dimensional packed arrays (vectors) that are ascending. It doesn't support multi-dimensional packed arrays.

garmin-mjames, 2021-01-14:
This was fixed in Verilator master, see https://github.com/verilator/verilator/issues/2696.

closed 2021-01-14 by garmin-mjames