cocotb/cocotb #2291: Scheduler._task_context performance

garmin-mjames, 2020-12-15:
While looking at @themperek's profiling log in #961, I noticed that `Scheduler._task_context` is taking a lot of time.

Investigating a little, the [contextlib docs](https://docs.python.org/3.5/library/contextlib.html?highlight=contextlib#contextlib.contextmanager) details say that using the `@contextmanager` decorator creates a new one-shot context manager on each call of the function, which involves some extra layers of wrappers due to the generic nature of **contextlib**.

This is used every time a task is scheduled. Investigate possible performance improvements from re-writing task context manager as a simpler dedicated context manager class.

ktbarrett, 2020-12-15:
Is it not possible to reuse the one-shot object? We could try just dropping a `@cache` decorator in there an see if that helps.

garmin-mjames, 2020-12-15:
We need a new context manager each time in order to pass in the new task.

It might be faster to use a stack and push/pop the current task. It's worth investigating performance on the scheduling hot path.

ktbarrett, 2020-12-15:
That sucks. The fact we have to have a stack of scheduled tasks also sucks. You are suggesting we have a single context object that when `__enter__`ing we push and `__exit__`ing we pop so we aren't continuously making new context manager objects?

eric-wieser, 2020-12-15:
Are you sure this isn't a red herring? `_task_context` might be showing up as also including all the time where the slow code is _within_ the context.

garmin-mjames, 2020-12-15:
Here are the relevant lines for `Scheduler.schedule` from the log. I'm not very familiar with `pprofile`.

```
Line #|      Hits|         Time| Time per hit|      %|Source code
------+----------+-------------+-------------+-------+-----------
   794|    337112|      3.14394|  9.32609e-06|  1.89%|        with self._task_context(coroutine):
(call)|    337112|      5.76703|  1.71072e-05|  3.47%|# /home/themperek/miniconda/lib/python3.7/contextlib.py:237 helper
(call)|    337112|      5.59323|  1.65916e-05|  3.37%|# /home/themperek/miniconda/lib/python3.7/contextlib.py:107 __enter__
```
And also it shows **contextlib.py** as using ~8% of the simulation time:
```
File: /home/themperek/miniconda/lib/python3.7/contextlib.py
File duration: 13.9257s (8.39%)
```

ktbarrett, 2020-12-15:
It is being executed on every scheduling operation. It looks like `contextlib overhead is 0.5s/35s of the sim, with the body of the manager taking 2.5s/35s. While it's maybe not that contextlib is slow, there could be a more efficient implementation.

closed 2023-11-08 by marlonjames