cocotb/cocotb #2294: Readback of a signal with length 0 (VHDL2008)

Botnic, 2020-12-16:
When I read back a signal of length 0, the conversion fails. 
I attached a minimal example. If the cntrlwidth is set to >0 it works. Signals with length 0 can for example be assigned to unused signals in buses. I expected the conversion to return 0.

```
Traceback (most recent call last):
#            File "binaryvalue_bug/binaryvaluetest.py", line 49, in test_cntrl
#            assert dut.Cntrl_in.value.integer == dut.Cntrl_out.value.integer
#            File "cocotb/cocotb/binary.py", line 294, in integer
#            return self._convert_from[self.binaryRepresentation](self._str)
#            File "cocotb/cocotb/binary.py", line 200, in _convert_from_unsigned
#            return int(resolve(x), 2)
#            ValueError: invalid literal for int() with base 2: ''

```
[binaryValue.zip](https://github.com/cocotb/cocotb/files/5701238/binaryValue.zip)

Running on ModelSim DE version 2020.4 2020.10 (32bit)
Running tests with cocotb v1.5.0.dev0
OS: CentOS 64bit

alexforencich, 2020-12-16:
Ha, I was in the process of typing up #2295 when you opened this issue.

alexforencich, 2020-12-16:
Interesting that this raises a ValueError, seems like this may just be a side-effect of how binaryvalue is currently implemented.

ktbarrett, 2020-12-16:
So, this issue was reported recently against cocotbext-axi: alexforencich/cocotbext-axi#8

Apparently, in VHDL, it is possible to have signals with length 0, and while cocotb can read such signals, converting the value to integer fails.

This is not necessarily a major problem on its own, but it becomes more annoying when said signal gets attached to a bus, as now one element in the bus cannot be converted to an integer.

It's unclear what the most appropriate fix would be. Personally, I think a zero-length signal should convert to 0. But perhaps there are some good reasons for it to not convert at all. Another possible fix would be to add a check to the bus module to throw out zero length signals, possibly based on an argument to __init__. Perhaps another option is to manually delete zero-length signals from the bus object after it is initialized, but IMO this is a bit of a hack and would require poking around in the internal state of the bus object.

Originally posted by @alexforencich in https://github.com/cocotb/cocotb/issues/2295#issue-768599978

ktbarrett, 2020-12-16:
The current behavior makes sense to me.

`BinaryValue` can handle 0-length signals, and I assume using the `.binstr` property on that BinaryValue works? It's only when you attempt to cast it to an integer with `int()` or use the `.integer` and `.signed_integer` properties that it throws an exception? You have no value and you are casting it to a type that cannot represent a lack of value, so it seems logical to throw. `0` may be convenient, but shadows the issue of the user probably doing the wrong thing.

I think the user or Bus type should handle these cases.

eric-wieser, 2020-12-16:
I think such a signal should read as zero. The only reason this fails right now is because `int(str)` is designed for interactive use, and a user entering an empty string is likely a mistake. Compare to `int.from_bytes`:
```
>>> int.from_bytes(b'', byteorder='little')
0
```

> you are casting it to a type that cannot represent a lack of value

A zero-bit signal is not the `empty` type with no values, it's the `unit` type with exactly one value.

ktbarrett, 2020-12-16:
> Compare to int.from_bytes, where a string of zero bytes is happily converted to zero

Seems inconsistent to me. But no good programming language is complete until it has a myriad of inconsistent interfaces that everyone has to remember.

> A zero-bit signal is not the empty type with no values, it's the unit type with exactly one value.

Right, and there is exactly 1 BinaryValue value to represent this state. For `0` to make sense the following would also have to work, otherwise the behavior is asymmetric.

```python
dut.empty_signal <= 0
dut.empty_signal <= BinaryValue(value=0, n_bits=0)
```

eric-wieser, 2020-12-16:
Yes, I think both of those should be legal too.

ktbarrett, 2020-12-16:
Well as long as it's consistent I guess I'm okay with it too. @Botnic @alexforencich Could one of you make the changes with a test for this case?

benreynwar, 2020-12-16:
I like this change.  I occasionally have ports that are indexing something, and when there is only one item to index they have a width of 0.  Currently I need extra logic in the tests for this special case, but with this change I'll be able to remove it.

ktbarrett, 2021-02-22:
What does the simulator do when writing to null arrays? Is that permitted according to the spec?

Botnic, 2021-02-23:
I can test on Modelsim only: 
Modelsim shows zero bit signals as ``-No Data-`` and does not allow any value forcing:
``** UI-Msg: (vsim-3455) Can force arrays only if they are one-dimensional arrays of character enums.``

eric-wieser, 2021-02-23:
How would you even write a null array with cocotb? Array writing is filled through python iteration, isn't it?

PR #2419 (2021-03-06): Add changes in BinaryValue to support zero vectors
Add changes in BinaryValue class to handle VHDL zero vectors better.
With this changes, a zero vector reads as 0 instead of throwing an exception.

As a beginner in contributing to your project, I do not expect this pull request to be complete or ready to be merged. I hope you can guide me to a successful contribution. Thank you.

closes #2294
closed 2021-03-06 by ktbarrett