cocotb/cocotb #2295: Zero-length signals and buses

alexforencich, 2020-12-16:
So, this issue was reported recently against cocotbext-axi: https://github.com/alexforencich/cocotbext-axi/issues/8

Apparently, in VHDL, it is possible to have signals with length 0, and while cocotb can read such signals, converting the value to integer fails.

It's unclear what the most appropriate fix would be.  Personally, I think a zero-length signal should convert to 0.  But perhaps there are some good reasons for it to not convert at all.  Another possible fix would be to add a check to the bus module to throw out zero length signals, possibly based on an argument to `__init__`.  Perhaps another option is to manually delete zero-length signals from the bus object after it is initialized, but IMO this is a bit of a hack and would require poking around in the internal state of the bus object.

Botnic, 2020-12-16:
Linked to #2294

ktbarrett, 2020-12-16:
Consolidating this with #2294 since they are about the same problem. He was first ;)

closed 2020-12-16 by ktbarrett