cocotb/cocotb #2296: Fail on error?

toddstrader, 2020-12-16:
Is there a way to make tests fail if, say, error or critical messages are logged?  I haven't found anything immediately and am reading up on the Python logging module, but wondered if this had been discussed previously (haven't found it in issues yet either).

ktbarrett, 2020-12-16:
Who is creating the error messages you want to fail on?

Failing when an error-level message occurs _by default_ is surprising behavior to most people. And unnecessary when the user can use `assert` to fail the test if they want to. You seem to be implying some thing outside of your control is creating an error that you think should be a failure (there are plenty of those cases in the C++ code).

If you want this behavior you can add a handler to the cocotb logger to call `cocotb.scheduler.finish_test` with an exception (I suggest `cocotb.result.TestFailure` or `AssertionError`). The cocotb logger is a standard Python logger, you can get more details [here](https://docs.python.org/3/library/logging.html) on how to define custom log handlers and how to register them.

toddstrader, 2020-12-16:
Nope, this is for errors that our test code logs.  Understood this is not the current behavior of cocotb, nor would it be reasonable to change it.  But I'm wondering if there's a way to add this as a global option to cocotb or a way to add hooks so that we could perform this linkage outside of cocotb.

I'll mess around with `logging` Handlers and see what I can come up with.

toddstrader, 2020-12-17:
Is this supposed to make the test fail?
```
$ git diff
diff --git a/examples/simple_dff/test_dff.py b/examples/simple_dff/test_dff.py
index d108ff4b..bd51356d 100644
--- a/examples/simple_dff/test_dff.py
+++ b/examples/simple_dff/test_dff.py
@@ -5,6 +5,7 @@ import random
 import cocotb
 from cocotb.clock import Clock
 from cocotb.triggers import FallingEdge
+import cocotb.result
 
 
 @cocotb.test()
@@ -20,3 +21,4 @@ async def test_dff_simple(dut):
         dut.d <= val  # Assign the random value val to the input port d
         await FallingEdge(dut.clk)
         assert dut.q == val, "output q was incorrect on the {}th cycle".format(i)
+    cocotb.scheduler.finish_test(cocotb.result.TestFailure)
```
Because I'm running this and `test_dff_simple` still passes (both in `_log_test_summary` and in `results.xml`).  I also tried `AssertionError` but that didn't change the results.

I'm still tracing this through but I'm seeing that a `RunningTest` gets its `_outcome` set to `Error(<class 'cocotb.result.TestFailure'>)`.  So far, that sounds like what I'm trying to do.

ktbarrett, 2020-12-17:
Ahhh, sorry for recommending what I did. The `finish_test` function only works when called from an external process; I forgot about that issue. This is due to a limitation in the implementation of the scheduler. The only way to fail a test while the test is running is by throwing a `TestComplete` exception or `assert`ing. You could use that instead, but it would fail to fail the test if an error message is logged in C++. I'm not sure if there is an easy way to accomplish what you are looking for.

eric-wieser, 2020-12-18:
> Nope, this is for errors that our test code logs.

Can your testcase use `raise RuntimeError("soemthing is wrong")` instead of `log.error("something is wrong")`?

eric-wieser, 2020-12-18:
Otherwise, you should be able to do something like [this](https://stackoverflow.com/a/48201163/102441):
```
class ExitOnExceptionHandler(logging.StreamHandler):
    def emit(self, record):
        super().emit(record)
        if record.levelno in (logging.ERROR, logging.CRITICAL):
            raise RuntimeError("Logging reported an error")

logging.getLogger("your_test_logger").addHandler(ExitOnExceptionHandler())
```

ktbarrett, 2020-12-18:
The unfortunate thing is that is throwing doesn't work if the log occurs in C.

closed 2020-12-21 by toddstrader

eric-wieser, 2020-12-21:
@toddstrader, are you writing your own C code that calls the GPI logging macros? Or is @ktbarrett's comment not relevant to you?

toddstrader, 2020-12-21:
No, we're only using the Python side of cocotb logging.  As discussed here and in the PR, wrapping `cocotb.test` and then adding an additional check forbidding the unwrapped `cocotb.test` should be sufficient.

However, I do get a meaningless stack trace when tests fail:
```
 36063.07ns ERROR    cocotb.regression                         regression.py:399  in _score_test                     Test Failed: test_invalid_instruction (result was TestFailure)                                
                                                                                                                     Traceback (most recent call last):                                                            
```

It only shows my test wrapper code, which of course makes sense since I'm deferring failure.  This doesn't block any of my current efforts since it is merely annoying, but I took a look at `_score_test()` to see what's happening here.  Would you accept a PR which adds a specialization of `TestFailure` which does not log the stack trace?

eric-wieser, 2020-12-21:
Can you show a few more lines of output so that I can see what you're actually annoyed about? It might be that simply raising `TestFailure` instead of `RuntimeError` is enough to make the problem go away?

toddstrader, 2020-12-21:
I am raising `TestFailure`, but `RuntimeError` produces the same result.  I believe the issue is that there's no path in `_score_test()` which does not expect a test failure that also does not print the traceback.

```
 36063.07ns ERROR    cocotb.regression                         regression.py:399  in _score_test                     Test Failed: test_invalid_instruction (result was TestFailure)                                
                                                                                                                     Traceback (most recent call last):                                                            
                                                                                                                       File "/usr/scratch/local/devs/dev-4/fpga/library/verification/cocohrt.py", line 42, in f    
                                                                                                                         handler.check_errors()                                                                    
                                                                                                                       File "/usr/scratch/local/devs/dev-4/fpga/library/verification/cocohrt.py", line 27, in check_errors
                                                                                                                         raise TestFailure                                                                         
                                                                                                                     cocotb.result.TestFailure                                                                     
```

eric-wieser, 2020-12-21:
Ah, I'd forgotten we now show `exc_info` for all errors:

https://github.com/cocotb/cocotb/blob/4eaf429c67739886fab45afd647b4eeb37436694/cocotb/regression.py#L399

You could probably fix this by registering a custom log handler that ignores custom exception types.

toddstrader, 2020-12-21:
Are you saying that I would remove the `ErrorLogHandler` that cocotb installs and add my own (which potentially wraps the cocotb one)?  I can clearly add multiple handlers, but as far as I understand the log records are broadcast to each of the handlers.  So simply adding a new handler won't help me get rid of this traceback.

I believe that the `ErrorLogHandler` is `addHandler()`'ed in `_start_test()` and `removeHandler()`'ed in `handle_result()`.  I'm not sure how I could remote the cocotb handler and install my own (without modifications to cocotb).

eric-wieser, 2020-12-21:
I wasn't thinking about the `ErrorLogHandler` at all. At any rate, that only affects the logger called `cocotb`, and if you use your a logger of a different name you'll be free to customize it