cocotb/cocotb #2302: Simulator-indepdent bit-slicing, shadow signals, and splitting concatenated buses

alexforencich, 2020-12-23:
What I would like to be able to do in cocotb is basically the equivalent of making a bunch of verilog `wires` and doing something similar to this:

    wire [7:0] port1;
    wire [7:0] port2;
    wire [7:0] port3;
    
    dut dut_inst (
        .port({port3, port2, port1})
    );
    
    model model_inst_1 (
        .port(port1)
    );
    
    model model_inst_2 (
        .port(port2)
    );
    
    model model_inst_3 (
        .port(port3)
    );

Except `dut` would be running in the simulator, and the `model`s would be cocotb simulation models.  The use case for this is for testing modules that use parametrized concatenated ports, such as: https://github.com/alexforencich/verilog-axi/blob/master/rtl/axil_interconnect.v :

    input  wire [S_COUNT*ADDR_WIDTH-1:0]  s_axil_awaddr,
    input  wire [S_COUNT*3-1:0]           s_axil_awprot,
    input  wire [S_COUNT-1:0]             s_axil_awvalid,
    output wire [S_COUNT-1:0]             s_axil_awready,
    input  wire [S_COUNT*DATA_WIDTH-1:0]  s_axil_wdata,
    input  wire [S_COUNT*STRB_WIDTH-1:0]  s_axil_wstrb,
    input  wire [S_COUNT-1:0]             s_axil_wvalid,
    output wire [S_COUNT-1:0]             s_axil_wready,
    output wire [S_COUNT*2-1:0]           s_axil_bresp,
    output wire [S_COUNT-1:0]             s_axil_bvalid,
    input  wire [S_COUNT-1:0]             s_axil_bready,
    input  wire [S_COUNT*ADDR_WIDTH-1:0]  s_axil_araddr,
    input  wire [S_COUNT*3-1:0]           s_axil_arprot,
    input  wire [S_COUNT-1:0]             s_axil_arvalid,
    output wire [S_COUNT-1:0]             s_axil_arready,
    output wire [S_COUNT*DATA_WIDTH-1:0]  s_axil_rdata,
    output wire [S_COUNT*2-1:0]           s_axil_rresp,
    output wire [S_COUNT-1:0]             s_axil_rvalid,
    input  wire [S_COUNT-1:0]             s_axil_rready,

It's a little cumbersome, but in MyHDL it is possible to create "shadow" signals that can either follow a portion of some other signal, or be formed from multiple signals concatenated together.  So, to connect this port to multiple simulation models, you first create the shadow signals:

https://github.com/alexforencich/verilog-axi/blob/master/tb/test_axil_interconnect_4x4.py#L65-L144

and then you can just loop over those to connect the models:

https://github.com/alexforencich/verilog-axi/blob/master/tb/test_axil_interconnect_4x4.py#L146-L220

In cocotb, it seems that there is currently no way to do this, so I resurrected some old wrapper generator scripts that I wrote years ago but never really used: https://github.com/alexforencich/verilog-axi/blob/master/rtl/axil_interconnect_wrap.py .  This basically does the concatenation on the HDL sides, and presents individual ports that can be accessed from cocotb.  Making the connections is very reasonable:

    self.axil_master = [AxiLiteMaster(dut, f"s{k:02d}_axil", dut.clk, dut.rst) for k in range(s_count)]
    self.axil_ram = [AxiLiteRam(dut, f"m{k:02d}_axil", dut.clk, dut.rst, size=2**16) for k in range(m_count)]

but writing and maintaining the wrapper generator is annoying if it's only used for testing the module.  

Ideally, I would like to be able to do something like this, without using the generated wrapper:

    self.axil_master = [AxiLiteMaster(seg, "s_axil", dut.clk, dut.rst) for seg in bus_split(dut, "s_axil")]
    self.axil_ram = [AxiLiteRam(seg, f"m_axil", dut.clk, dut.rst, size=2**16) for seg in bus_split(dut, "m_axil")]

where `bus_split` would match all signals with the prefix, determine the number of splits (probably pick the narrowest signal width, but there could be some variations on that) and then return a list of objects that contain shadow signals that either drive or follow the corresponding subset of bits in the original signal.  

These shadow signals could come from bit slicing the signals from the simulator, like so:

    port1 = dut.port[0:8]
    port2 = dut.port[8:16]
    port3 = dut.port[16:24]

Or possibly by direct instantiation, possibly with explicit direction indication:

    port1 = ShadowSignal(dut.port, 0, 8)
    port2 = ShadowSignal(dut.port, 8, 16)
    port3 = ShadowSignal(dut.port, 16, 24)
    
    port1 = ShadowSignalRead(dut.port, 0, 8)
    port2 = ShadowSignalRead(dut.port, 8, 16)
    port3 = ShadowSignalRead(dut.port, 16, 24)
    
    port1 = ShadowSignalWrite(dut.port, 0, 8)
    port2 = ShadowSignalWrite(dut.port, 8, 16)
    port3 = ShadowSignalWrite(dut.port, 16, 24)

Now, I don't think VPI/VHPI support this directly.  But, I think it should be possible to emulate this.  Reading from one of these shadow signals would simply perform a full-width read of the original signal, and then return the appropriate bit-sliced version.  Writing a shadow signal is potentially a bit more complex.  For simulators that support bit-access, then the write can be split up into the individual bits and written that way.  But this is likely inefficient, especially for wide buses.  However, as many of these signals will be unidirectional, another option is to coalesce partial writes into full-width writes and write the entire signal, including bits that did not change in that timestep.

Ideally, this would be implemented at the GPI layer so that it can be utilized by simulation models that are implemented in C/C++ and use GPI directly.

eric-wieser, 2020-12-23:
> These shadow signals could come from bit slicing the signals from the simulator, like so:

This seems like quite a reasonable suggestion to me, but we'd need to make quite clear the difference between extracting the value of a bit slice and extracting a "pointer" or "view". I think your suggestion is roughly `signal.value[i:j]` and `signal[i:j]` respectively for those spellings, but we probably want to keep the shortest one for the most common use case - perhaps one `signal.view[i:j]`, `view(signal)[i:j]` or `view[i:j](signal)` would be better for the shadowing version (swapping `view` for a name of your choice).

* `signal.value[i:j]` vs `signal[i:j]` (what your suggestion seems to be
* `signal[i:j]` vs `signal.view[i:j]`
* `signal[i:j]` vs `view(signal)[i:j]`

Regarding `bus_split`, I think that sounds like something to just provide as an example in documentation, rather than including in cocotb - I'd be wary of doing anything heuristic in the API if we can avoid it.

alexforencich, 2020-12-23:
right, `bus_split` would be a helper method, where it is implemented is somewhat immaterial (although IMHO, if there is a `bus`, then `bus_split` could go in the same file), the important part is getting to the point where it is possible to write such a routine.

alexforencich, 2021-01-02:
Extension to this idea: shadow signals for other basic manipulations of simulator signals, specifically bitwise invert.  This can be useful if you want to invert a reset signal or something similar.  

Possible ways to implement this on the python side:

Add a global `inverted()` method.  In this case, you can get an inverted signal like so:

    nrst = inverted(dut.rst)

Add an `inverted()` method to the signal.  In this case, you can get an inverted signal like so:

    nrst = dut.rst.inverted()

Implement `__invert__()` on the signal.  In this case, you can get an inverted signal like so:

    nrst = ~dut.rst

Looks like there is no way to override `not` as this casts to `bool` and then inverts that.

ktbarrett, 2021-06-11:
I think another viable use case for user created signals is for flow control: How do you create a Driver that can handle flow control without building the concept of flow control into the driver and describing the semantics? Condition control signals you pass to the driver by combining the HDL signal with a pseudo-signal.

alexforencich, 2021-06-12:
I guess with that one you would write a coroutine to implement the flow control semantics and then exert backpressure against a driver model in some way, perhaps directly though python.  But yes, this would require some explicit support in the driver model, as opposed to ANDing something with both the ready and valid signals.  But, we would also have to figure out how to rework the bus object to "pick off" the flow control signals.  I'm planning on making a replacement for the bus object anyway where perhaps we could take something like this into consideration.

ktbarrett, 2021-06-12:
@alexforencich I think this will need support in cocotb proper to supporting bit-sliced handles with edge triggers. Might as well wring the full functionality out of such a feature.

> I'm planning on making a replacement for the bus object

Could you list the issues with the Bus object? I'm curious of the opinions of someone who has used it extensively since I have not.