cocotb/cocotb #2303: Using array_idx to create Drivers fails in Verilator

phhorrein, 2020-12-24:
cocotb version: master
OS: Linux Ubuntu 64bits
simulator: verilator v4.106
Python: 3.6

I'm trying to create multiple drivers on unpacked buses. Here is how I do it: 
```python
for bank in range(banks):
    self.in_bus.append(MyBusDriver(dut, "in", dut.clk, array_idx=bank))
    self.out_bus.append(MyOutMonitor(dut, "out", dut.clk, array_idx=bank))
```

The driver is defined with a "rdy" signal to it (among other signals). 

Running this in questa, with logging level set to DEBUG, things work out correctly:
```
#      0.00ns DEBUG    cocotb.my_wrapper                               bus.py:107  in _add_signal                     Signal name in_rdy[1]
```

However, this fails in Verilator (I guess that the VPI interface is different).

```
...cocotb/handle.py", line 316, in __getattr__
   raise AttributeError("%s contains no object named %s" % (self._name, name))
AttributeError: my_wrapper contains no object named in_rdy[0]
```

 I'm trying to dig, in order to fix this, but I'm still new to cocotb (if not to VPI). Has anyone an idea on where I should be looking at to fix this?

phhorrein, 2020-12-24:
I created a small test, which works with Icarus and Questa, but not with Verilator, to demonstrate what happens. 
[array_driver_bug.zip](https://github.com/cocotb/cocotb/files/5739575/array_driver_bug.zip)

phhorrein, 2020-12-24:
When I add some prints (sorry, not the prettiest way to debug, but in this case it's fast :)), I see a first bug: 
 ```python
    cocotb.log.warning("Data is: {}, of type {}".format(dut.in_data, type(dut.in_data)))
    cocotb.log.warning("Data 0 is: {}, of type {}".format(dut.in_data[0].value, type(dut.in_data[0])))
    cocotb.log.warning("Valid is: {}, of type {}".format(dut.in_valid, type(dut.in_valid)))
    cocotb.log.warning("Valid 0 is: {}, of type {}".format(dut.in_valid[0].value, type(dut.in_valid[0])))
```

Running with Icarus, I get: 
```
Data is: packed_bug.in_data, of type <class 'cocotb.handle.NonHierarchyIndexableObject'>
Data 0 is: zzzzzzzz, of type <class 'cocotb.handle.ModifiableObject'>
Valid is: packed_bug.in_valid, of type <class 'cocotb.handle.NonHierarchyIndexableObject'>
Valid 0 is: z, of type <class 'cocotb.handle.ModifiableObject'>
```

However, running with Verilator: 
```
Data is: packed_bug.in_data, of type <class 'cocotb.handle.NonHierarchyIndexableObject'>
Data 0 is: 00000000, of type <class 'cocotb.handle.ModifiableObject'>
Valid is: 00, of type <class 'cocotb.handle.ModifiableObject'>
in gpi_get_handle_by_index         Failed to find a handle at index 0 via any registered implementation
```
which means that the valid (an unpacked array of 2 logic) is seen as a packed array...

But removing the valid from the driver does not remove the error, So I guess these are two different issues.

phhorrein, 2020-12-24:
I tried a workaround, which fixes the first issue. I'm unsure on how the second issue can be solved, it seems that Verilator creates this as a Net as it is 1-dimension.

ktbarrett, 2020-12-24:
@phhorrein You should try accessing these signals without using the Bus to confirm this as an issue with the bus. I have a feeling that searching for the name `array[0]` ([here](https://github.com/cocotb/cocotb/blob/9b442d07870751ff656eb1ed53d24a3d806e9610/cocotb/bus.py#L86)) may be defeating the code we have which sets up pseudo-regions.

phhorrein, 2021-01-08:
Fixed by #2304

closed 2021-01-08 by phhorrein