cocotb/cocotb #2308: Float value in testbench

kgokarn, 2020-12-29:
Hi,

I am a new user of cocotb. I have used it on simpler adders and flipflops and they work fine for integer values or assignment of zeros and ones.  But when trying to instantiate a Verilog code for 5X5 image, I get an error for float values.

               dut.i_activation_0_0 <= new_x[0, 0]
                dut.i_activation_0_1 <= new_x[0, 1]
                dut.i_activation_0_2 <= new_x[0, 2]
                dut.i_activation_0_3 <= new_x[0, 3]
                dut.i_activation_0_4 <= new_x[0, 4]
                dut.i_activation_1_0 <= new_x[1, 0]
                dut.i_activation_1_1 <= new_x[1, 1]
                dut.i_activation_1_2 <= new_x[1, 2]
                dut.i_activation_1_3 <= new_x[1, 3]
                dut.i_activation_1_4 <= new_x[1, 4]
                dut.i_activation_2_0 <= new_x[2, 0]
                dut.i_activation_2_1 <= new_x[2, 1]
                dut.i_activation_2_2 <= new_x[2, 2]
                dut.i_activation_2_3 <= new_x[2, 3]
                dut.i_activation_2_4 <= new_x[2, 4]
                dut.i_activation_3_0 <= new_x[3, 0]
                dut.i_activation_3_1 <= new_x[3, 1]
                dut.i_activation_3_2 <= new_x[3, 2]
                dut.i_activation_3_3 <= new_x[3, 3]
                dut.i_activation_3_4 <= new_x[3, 4]
                dut.i_activation_4_0 <= new_x[4, 0]
                dut.i_activation_4_1 <= new_x[4, 1]
                dut.i_activation_4_2 <= new_x[4, 2]
                dut.i_activation_4_3 <= new_x[4, 3]
                dut.i_activation_4_4 <= new_x[4, 4]

new_x matrix contains float64 values which I have to assign to input port of verilog code, which are on the left hand side. Is there any solution for this or am I missing something?

Thanks in advance

ktbarrett, 2020-12-29:
Instead of ignoring the template can you please fill it out? What simulator, simulator version, OS, and cocotb version are you using? Post the full trace back or error message.

kgokarn, 2020-12-30:
Sorry!

cocotb version is 1.4.0
I am running on Ubunu 18.04LTS, 64-bit OS
I am using Icarus Verilog Simulator 10.1
Python 3.6 (System)

Verilog Port Information:

module top(
    input wire clk,
    input wire reset,
    input wire signed [11:0] i_activation_0_0,
    input wire signed [11:0] i_activation_0_1,
    input wire signed [11:0] i_activation_0_2,
    input wire signed [11:0] i_activation_0_3,
    input wire signed [11:0] i_activation_0_4,
    input wire signed [11:0] i_activation_1_0,
    input wire signed [11:0] i_activation_1_1,
    input wire signed [11:0] i_activation_1_2,
    input wire signed [11:0] i_activation_1_3,
    input wire signed [11:0] i_activation_1_4,
    input wire signed [11:0] i_activation_2_0,
    input wire signed [11:0] i_activation_2_1,
    input wire signed [11:0] i_activation_2_2,
    input wire signed [11:0] i_activation_2_3,
    input wire signed [11:0] i_activation_2_4,
    input wire signed [11:0] i_activation_3_0,
    input wire signed [11:0] i_activation_3_1,
    input wire signed [11:0] i_activation_3_2,
    input wire signed [11:0] i_activation_3_3,
    input wire signed [11:0] i_activation_3_4,
    input wire signed [11:0] i_activation_4_0,
    input wire signed [11:0] i_activation_4_1,
    input wire signed [11:0] i_activation_4_2,
    input wire signed [11:0] i_activation_4_3,
    input wire signed [11:0] i_activation_4_4,
    input wire signed [11:0] i_weight_0_0_0,
    input wire signed [11:0] i_weight_0_0_1,
    input wire signed [11:0] i_weight_0_0_2,
    input wire signed [11:0] i_weight_0_0_3,
    input wire signed [11:0] i_weight_0_1_0,
    input wire signed [11:0] i_weight_0_1_1,
    input wire signed [11:0] i_weight_0_1_2,
    input wire signed [11:0] i_weight_0_1_3,
    input wire signed [11:0] i_weight_0_2_0,
    input wire signed [11:0] i_weight_0_2_1,
    input wire signed [11:0] i_weight_0_2_2,
    input wire signed [11:0] i_weight_0_2_3,
    input wire signed [11:0] i_weight_0_3_0,
    input wire signed [11:0] i_weight_0_3_1,
    input wire signed [11:0] i_weight_0_3_2,
    input wire signed [11:0] i_weight_0_3_3,
    input wire signed [11:0] i_weight_1_0_0,
    input wire signed [11:0] i_weight_1_0_1,
    input wire signed [11:0] i_weight_1_0_2,
    input wire signed [11:0] i_weight_1_0_3,
    input wire signed [11:0] i_weight_1_1_0,
    input wire signed [11:0] i_weight_1_1_1,
    input wire signed [11:0] i_weight_1_1_2,
    input wire signed [11:0] i_weight_1_1_3,
    input wire signed [11:0] i_weight_1_2_0,
    input wire signed [11:0] i_weight_1_2_1,
    input wire signed [11:0] i_weight_1_2_2,
    input wire signed [11:0] i_weight_1_2_3,
    input wire signed [11:0] i_weight_1_3_0,
    input wire signed [11:0] i_weight_1_3_1,
    input wire signed [11:0] i_weight_1_3_2,
    input wire signed [11:0] i_weight_1_3_3,
    input wire signed [11:0] i_weight_2_0_0,
[... 692 lines cut ...]
                layer2[count, count2, 0] = dut.o_output_0
                layer2[count, count2, 1] = dut.o_output_1
                layer2[count, count2, 2] = dut.o_output_2
                layer2[count, count2, 3] = dut.o_output_3
                layer2[count, count2, 4] = dut.o_output_4
                layer2[count, count2, 5] = dut.o_output_5
                layer2[count, count2, 6] = dut.o_output_6
                layer2[count, count2, 7] = dut.o_output_7
                layer2[count, count2, 8]  = dut.o_output_8
                layer2[count, count2, 9] = dut.o_output_9
                layer2[count, count2, 10] = dut.o_output_10
                layer2[count, count2, 11] = dut.o_output_11
                layer2[count, count2, 12] = dut.o_output_12
                layer2[count, count2, 13] = dut.o_output_13
                layer2[count, count2, 14] = dut.o_output_14
                layer2[count, count2, 15] = dut.o_output_15
                layer2[count, count2, 16] = dut.o_output_16
                layer2[count, count2, 17] = dut.o_output_17
                layer2[count, count2, 18] = dut.o_output_18
                layer2[count, count2, 19] = dut.o_output_19

I know conversion from float64 to 12bit fixed point is not done here, but I am unable to get it to work for float. I have posted the error message in the attachment.

Thanks.
[image]

ktbarrett, 2020-12-30:
```
TypeError: Unsupported type for value assignment: <class 'numpy.float64'> (0.0)
```

The exception seems to be fairly clear to me. `numpy.float64` values are not supported in float assignments in cocotb. You will have to convert them to Python floats first. @eric-wieser may have a better answer, since he is very familiar with numpy, but Google tells me you can use the `.item()` method on numpy values to convert them into Python types.

Could you try something like the following?
```python
dut.i_weight_18_3_3 <= weights[3, 3, 18].item()
```

You also might be able to prevent the huge copy paste by leveraging Python:
```python
for dim_1 in range(4):
    for dim_2 in range(4):
        for dim_3 in range(20):
            name = f"i_weight_{dim_3}_{dim_1}_{dim_2}"
            handle = getattr(dut, name)
            handle.value = weights[dim1, dim2, dim3].item()
```

One last thing to mention is that Icarus 10.1 is a rather old release and does not fully support cocotb. You *may* have some issues with it, so just be aware of that.

kgokarn, 2020-12-30:
Hi, 

Thanks for the reply. I have tried item() earlier. Instead of "numpy.float64 values are not supported in float assignments in cocotb", I get "TypeError: Unsupported type for value assignment: <class 'float'> (0.0)" error. I will update Icarus to newer version and check, and also take into consideration to minimise the code.
I shall let you know if I have same problem with new version of Icarus Verilog.

ktbarrett, 2020-12-30:
Ah, then it is probably an issue with an old issue of Icarus then. Icarus 10.3 is 95% feature coverage and 11.0 is 99+% AFAIK. I recommend building Icarus from scratch from the `v11-branch` branch.

kgokarn, 2020-12-30:
I could compile source from V-11 branch, but they do not have steps to install on Ubuntu. It says they are working on it. With I being on 18.04, that leaves 10.1 the latest version of icarus on my computer.

ktbarrett, 2020-12-30:
>  but they do not have steps to install on Ubuntu

```
sudo make install
```
Should work.

kgokarn, 2020-12-30:
Sorry! I was trying to install from the root. Now iverilog 11.0 is installed, but the problem with float persists.

ktbarrett, 2020-12-30:
I guess I assumed you were trying to assign a Python float to a `real`/`shortreal` signal and didn't look through your dump. You are defining the signals as signed 12 bit values, but you are assigning 64 bit floats. The types don't match up. What are you trying to do exactly? How do you want the float values interpreted for assignment?

kgokarn, 2020-12-30:
The verilog 12 bit ports are signed fixed point numbers. I have a function that converts float64 to 12-bit signed fixed point.

from fxpmath import Fxp
def fix_pt(val):
    pi_fxp = Fxp(None, signed=True, n_word=12, n_frac=8)   
    pi_fxp.rounding = 'around'                             
    pi_fxp(val)     
   
Say, I am instantiating the port as "dut.i_activation_0_0 <= fix_pt(new_x[0, 0].item())", I will still get the error:    TypeError: Unsupported type for value assignment: <class 'NoneType'> (None).

So before posting I removed the function from the code as I believe the error is indeed coming from float. I tried using float in a very simple ripple carry adder too, to check my hypothesis. I got same error. Maybe I am not handling float correctly in the testbench.

ktbarrett, 2020-12-30:
You are not returning anything in that function. logic array types, like your `signed [11:0]` only accept assignment with `cocotb.binary.BinaryValue` and Python `int`s. You probably need the following:

```python
from fxpmath import Fxp

pi_fxp = Fxp(None, signed=True, n_word=12, n_frac=8)   # no need to recreate this type every time
pi_fxp.rounding = 'around'   

def fix_pt(val):                          
    return int(pi_fxp(val))
```

Then you can do `dut.i_activation_0_0 <= fix_pt(new_x[0, 0].item())`. I might even move `.item()` into `fix_pt` to prevent repeating it a bunch.

kgokarn, 2020-12-30:
Thanks, I will try this.  I shall keep you posted if this works!

kgokarn, 2020-12-30:
One more question. Should it be return int(pi_fxp(val)) or return int(pi_fxp(val).bin())?

ktbarrett, 2020-12-30:
I'm sorry I don't know anything about `fxpmath`, .`bin()` sounds promising.

ktbarrett, 2020-12-31:
@kgokarn Is your issue resolved?

kgokarn, 2021-01-01:
@ktbarrett Yes, it is solved. Sorry for responding late. 

Thanks for your help.

closed 2021-01-01 by themperek