cocotb/cocotb #2310: cannot find -lgpilog

catkira, 2020-12-29:
System: Windows 10 64 Bit, msys2

It works for me when I install it in conda, but not in msys2. I need msys2 because I want to use it together with verilator in windows.

Error message:

benja@LAPTOP-Benjamin MINGW64 /d/_shared_git/cocotb
$ pip install https://github.com/cocotb/cocotb/archive/master.zip --user
Collecting https://github.com/cocotb/cocotb/archive/master.zip
  Using cached https://github.com/cocotb/cocotb/archive/master.zip
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: cocotb
  Building wheel for cocotb (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python.exe /usr/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmpqdh7rgqk
       cwd: /tmp/pip-req-build-_i27e6ag
  Complete output (107 lines):
  Modelsim/Questa executable (vdbg) executable not found. No FLI interface will be available.
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8
  creating build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/ANSI.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/binary.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/bus.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/clock.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/config.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/decorators.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/handle.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/ipython_support.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/log.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/memdebug.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/outcomes.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/regression.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/result.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/scheduler.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/scoreboard.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/triggers.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/utils.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/wavedrom.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/xunit_reporter.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/_py_compat.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/_sim_versions.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/_version.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  copying cocotb/__init__.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  creating build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/drivers
  copying cocotb/drivers/amba.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/drivers
  copying cocotb/drivers/avalon.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/drivers
  copying cocotb/drivers/opb.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/drivers
  copying cocotb/drivers/xgmii.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/drivers
  copying cocotb/drivers/__init__.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/drivers
  creating build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/generators
  copying cocotb/generators/bit.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/generators
  copying cocotb/generators/byte.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/generators
  copying cocotb/generators/packet.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/generators
  copying cocotb/generators/__init__.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/generators
  creating build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/monitors
  copying cocotb/monitors/avalon.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/monitors
  copying cocotb/monitors/xgmii.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/monitors
  copying cocotb/monitors/__init__.py -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/monitors
  creating build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/share
  creating build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/share/makefiles
  copying cocotb/share/makefiles/Makefile.deprecations -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/share/makefiles
  copying cocotb/share/makefiles/Makefile.inc -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/share/makefiles
  copying cocotb/share/makefiles/Makefile.sim -> build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/share/makefiles
  creating build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/share/makefiles/simulators
[... 35 lines cut ...]
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libgpilog
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libgpilog/cocotb
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libgpilog/cocotb/share
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libgpilog/cocotb/share/lib
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libgpilog/cocotb/share/lib/gpi_log
  x86_64-pc-msys-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=x86-64 -mtune=generic -O2 -pipe -march=x86-64 -mtune=generic -O2 -pipe -march=x86-64 -mtune=generic -O2 -pipe -DGPILOG_EXPORTS= -D__STDC_FORMAT_MACROS= -Icocotb/share/include -I/usr/include/python3.8 -c cocotb/share/lib/gpi_log/gpi_logging.cpp -o build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libgpilog/cocotb/share/lib/gpi_log/gpi_logging.o -std=c++11 -fvisibility=hidden -fvisibility-inlines-hidden -Wall -Wextra -Wcast-qual -Wwrite-strings -Wconversion -Wnon-virtual-dtor -Woverloaded-virtual -flto
  creating build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs
  /usr/bin/g++ -shared -Wl,--enable-auto-image-base -pipe -L. -pipe -L. build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libgpilog/cocotb/share/lib/gpi_log/gpi_logging.o -Lbuild/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs -L/usr/lib/python3.8/config -L/usr/lib -lpython3.8 -o build/lib.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libgpilog.so -Wl,-rpath,$ORIGIN -flto
  building 'cocotb/libs/libcocotbutils' extension
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libcocotbutils
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libcocotbutils/cocotb
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libcocotbutils/cocotb/share
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libcocotbutils/cocotb/share/lib
  creating build/temp.mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libcocotbutils/cocotb/share/lib/utils
  x86_64-pc-msys-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=x86-64 -mtune=generic -O2 -pipe -march=x86-64 -mtune=generic -O2 -pipe -march=x86-64 -mtune=generic -O2 -pipe -DCOC [... cut ...] mingw64_nt-10.0-18363-3.1.7-340.x86_64-x86_64-3.8/cocotb/libs/libcocotbutils.so -Wl,-rpath,$ORIGIN -flto
  /usr/lib/gcc/x86_64-pc-msys/10.2.0/../../../../x86_64-pc-msys/bin/ld: cannot find -lgpilog
  collect2: error: ld returned 1 exit status
  error: command '/usr/bin/g++' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for cocotb
Failed to build cocotb
ERROR: Could not build wheels for cocotb which use PEP 517 and cannot be installed directly

ktbarrett, 2020-12-30:
> I need msys2 because I want to use it together with verilator in windows.

Are you attempting to use the prebuilt msys2 package for verilator? You can still accomplish this using conda by installing Verilator from source. We currently recommend and only test cocotb on Windows using conda, other Windows environments will have issues until we try to sort them out.

If you can find the probably we are happy to accept patches.

themperek, 2020-12-30:
Did you follow instructions from? https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#windows-installation-using-msys2
Especially point `2. Start mingw32/64.exe from MSYS2 installation directory.`

catkira, 2020-12-30:
I was using msys2 in a conda environment. But I think trying to get these things run in windows is a hopeless adventure. I am using WSL2 (windows subsystem for linux 2) now and everything runs very smooth. cocotb, verilator, iverilog - no problems. I recommend that you make this your main recommendation for people who want to get it running in windows.

themperek, 2020-12-30:
I cannot reproduce your issues if I follow https://github.com/cocotb/cocotb/wiki/Tier-2-Setup-Instructions#windows-installation-using-msys2

I will close the issues. Feel free to reopen if needed.

ktbarrett, 2020-12-30:
@catkira I will agree that WSL is the way to go if you can use it. Windows is not a very friendly development environment as soon as you step out of Microsoft's environment. Many more popular Python packages than ours have troubles with Windows as well.

catkira, 2020-12-30:
I switched to using WSL2, it works very nice, highly recommended for windows users.

closed 2020-12-30 by catkira

svenka3, 2022-10-09:
> 

I encountered exact same issue on a Windows 10, under MSYS (64 bit) and that "Point-2" was dot-on-target. Thanks a lot!