cocotb/cocotb #2317: Testbench instantiation in a loop

kgokarn, 2021-01-02:
Cocotb version : 1.4.0
OS : Ubuntu 18.04LTS/64bit
Simulator : Icarus Verilog 11.0
python 3.6

I have a python code for neural networks. One of the layers of neural network is a hardware. So I need to instantiate  the hardware from the python script. The instantiation of hardware is done within a loop using cocotb, however, I am not getting it to run the way I want.

    while(iter<25):
        ii=0
        count2 = 0
        while(ii<25):
            new_x = img[ii:ii+5,iter:iter+5]
            #print("count is:",count)
            #print("count2 is:",count2)
            print("iter before is",iter)
            print("ii before is",ii)

            
            
            
            @cocotb.test()
            async def test_cnn_hw(dut):
                clock = Clock(dut.clk, 10, units="ns")  # Create a 10ns period clock on port clk
                cocotb.fork(clock.start())  # Start the clock

                reset = 0
                
                print("iter before is",iter)
                print("ii before is",ii)

                await RisingEdge(dut.clk)
                dut.reset <= reset
                dut.i_activation_0_0 <= fix_pt(new_x[0, 0])
                dut.i_activation_0_1 <= fix_pt(new_x[0, 1])
                dut.i_activation_0_2 <= fix_pt(new_x[0, 2])
                dut.i_activation_0_3 <= fix_pt(new_x[0, 3])
                dut.i_activation_0_4 <= fix_pt(new_x[0, 4])
                dut.i_activation_1_0 <= fix_pt(new_x[1, 0])
                dut.i_activation_1_1 <= fix_pt(new_x[1, 1])
                dut.i_activation_1_2 <= fix_pt(new_x[1, 2])
                dut.i_activation_1_3 <= fix_pt(new_x[1, 3])
                dut.i_activation_1_4 <= fix_pt(new_x[1, 4])
                dut.i_activation_2_0 <= fix_pt(new_x[2, 0])
                dut.i_activation_2_1 <= fix_pt(new_x[2, 1])
                dut.i_activation_2_2 <= fix_pt(new_x[2, 2])
                dut.i_activation_2_3 <= fix_pt(new_x[2, 3])
                dut.i_activation_2_4 <= fix_pt(new_x[2, 4])
                dut.i_activation_3_0 <= fix_pt(new_x[3, 0])
                dut.i_activation_3_1 <= fix_pt(new_x[3, 1])
                dut.i_activation_3_2 <= fix_pt(new_x[3, 2])
                dut.i_activation_3_3 <= fix_pt(new_x[3, 3])
                dut.i_activation_3_4 <= fix_pt(new_x[3, 4])
                dut.i_activation_4_0 <= fix_pt(new_x[4, 0])
                dut.i_activation_4_1 <= fix_pt(new_x[4, 1])
                dut.i_activation_4_2 <= fix_pt(new_x[4, 2])
                dut.i_activation_4_3 <= fix_pt(new_x[4, 3])
                dut.i_activation_4_4 <= fix_pt(new_x[4, 4])
                .
                .
                .
                layer2[count, count2, 0] = dut.o_output_0
                layer2[count, count2, 1] = dut.o_output_1
                layer2[count, count2, 2] = dut.o_output_2
                layer2[count, count2, 3] = dut.o_output_3
                layer2[count, count2, 4] = dut.o_output_4
                layer2[count, count2, 5] = dut.o_output_5
                layer2[count, count2, 6] = dut.o_output_6
                layer2[count, count2, 7] = dut.o_output_7
                layer2[count, count2, 8]  = dut.o_output_8
                layer2[count, count2, 9] = dut.o_output_9
                layer2[count, count2, 10] = dut.o_output_10
                layer2[count, count2, 11] = dut.o_output_11
                layer2[count, count2, 12] = dut.o_output_12
                layer2[count, count2, 13] = dut.o_output_13
                layer2[count, count2, 14] = dut.o_output_14
                layer2[count, count2, 15] = dut.o_output_15
                layer2[count, count2, 16] = dut.o_output_16
                layer2[count, count2, 17] = dut.o_output_17
                layer2[count, count2, 18] = dut.o_output_18
                layer2[count, count2, 19] = dut.o_output_19
                
                            
            ii = ii + 2
            count2 = count2+1
        iter = iter + 2
        
        count=count+1

However, I am unable to print it for every iteration. If I print 'iter' or 'ii' after @cocotb.test(), it is always 26. So is my 'count' or 'count2' which is always 13. But it should vary between [0,12]. If I print them outside, I see the iterators and count variables vary between the intended values.

Is it possible to write a cocotb testbench the way I have written or should I take care of something?

Thanks in advance.

eric-wieser, 2021-01-02:
I don't really understand what you're trying to do, but what you're seeing is that python functions close over variables names not their values.

Using `async def test_cnn_hw(dut, ii=ii, iter=iter):` instead of what you have will solve that problem, but I suspect what you're trying to do is not sensible in its current form, as cocotb cannot instantiate hdl blocks.

kgokarn, 2021-01-02:
Thank you for your reply. I have the image in new_x. new_x is a numpy array of 5X5, and it is continuously written. I am trying to assign those values to verilog ports and then read the outcome from output port and save it in another numpy array. 

Can you tell me if something is wrong? I am very new to cocotb and still exploring things.

Thanks.

ktbarrett, 2021-01-03:
> So I need to instantiate the hardware from the python script

This is not possible. cocotb cannot instantiate hardware; only test hardware that was previously defined. If you need to instantiate a NN layer module multiple times, you will have to do it in Verilog/VHDL. cocotb can then be used to program the weights into each layer and run tests through it, which is what it seems like you are trying to do from that code snippet you posted.

kgokarn, 2021-01-03:
Yes, instantiation happens for modules within another module. Sorry for the poor use of words.

I have multiple layers in NN. Only one layer is in Verilog. The output from Verilog is again input to the other layers. In makefile, I have set python module to the NN python script. And from the python script, I am invoking cocotb functions.

   while(iter<25):
        ii=0
        count2 = 0
        while(ii<25):
            new_x = img[ii:ii+5,iter:iter+5]
            loop = asyncio.new_event_loop()
            asyncio.set_event_loop(loop)
            layer2 = loop.run_until_complete(tstbench.test_cnn_hw(new_x,weights,layer2,iter,ii,count,count2))     
            ii = ii + 2
            count2 = count2+1
        iter = iter + 2
        
        count=count+1

I have imported the python file of cocotb in this one. 

This is my cocotb testbench looks like:

import cocotb
from cocotb.clock import Clock
from cocotb.triggers import RisingEdge
import time
from fxpmath import Fxp
from rig.type_casts import fix_to_float

pi_fxp = Fxp(None, signed=True, n_word=12, n_frac=8)    
pi_fxp.rounding = 'around'   

def fix_pt(val):                          
    return int(pi_fxp(val).bin())

def fix2float(val):
    f2f= fix_to_float(True,12, 10)
    return f2f(val)

@cocotb.test()
async def test_cnn_hw(dut, new_x=new_x, weights=weights,layer2=layer2,iter=iter,ii=ii,count=count,count2=count2):
    clock = Clock(dut.clk, 10, units="ns")  # Create a 10ns period clock on port clk
    cocotb.fork(clock.start())  # Start the clock

    reset = 0

    print("iter before is", iter)
    print("ii before is", ii)

    await RisingEdge(dut.clk)
    dut.reset <= reset
    dut.i_activation_0_0 <= fix_pt(new_x[0, 0])
    dut.i_activation_0_1 <= fix_pt(new_x[0, 1])
    dut.i_activation_0_2 <= fix_pt(new_x[0, 2])
    dut.i_activation_0_3 <= fix_pt(new_x[0, 3])
    dut.i_activation_0_4 <= fix_pt(new_x[0, 4])
    dut.i_activation_1_0 <= fix_pt(new_x[1, 0])
    dut.i_activation_1_1 <= fix_pt(new_x[1, 1])
    dut.i_activation_1_2 <= fix_pt(new_x[1, 2])
    dut.i_activation_1_3 <= fix_pt(new_x[1, 3])
    dut.i_activation_1_4 <= fix_pt(new_x[1, 4])
    dut.i_activation_2_0 <= fix_pt(new_x[2, 0])
    dut.i_activation_2_1 <= fix_pt(new_x[2, 1])
    dut.i_activation_2_2 <= fix_pt(new_x[2, 2])
    dut.i_activation_2_3 <= fix_pt(new_x[2, 3])
    dut.i_activation_2_4 <= fix_pt(new_x[2, 4])
    dut.i_activation_3_0 <= fix_pt(new_x[3, 0])
    dut.i_activation_3_1 <= fix_pt(new_x[3, 1])
    dut.i_activation_3_2 <= fix_pt(new_x[3, 2])
    dut.i_activation_3_3 <= fix_pt(new_x[3, 3])
[... 352 lines cut ...]
    layer2[count, count2, 3] = fix2float(dut.o_output_3)
    layer2[count, count2, 4] = fix2float(dut.o_output_4)
    layer2[count, count2, 5] = fix2float(dut.o_output_5)
    layer2[count, count2, 6] = fix2float(dut.o_output_6)
    layer2[count, count2, 7] = fix2float(dut.o_output_7)
    layer2[count, count2, 8] = fix2float(dut.o_output_8)
    layer2[count, count2, 9] = fix2float(dut.o_output_9)
    layer2[count, count2, 10] = fix2float(dut.o_output_10)
    layer2[count, count2, 11] = fix2float(dut.o_output_11)
    layer2[count, count2, 12] = fix2float(dut.o_output_12)
    layer2[count, count2, 13] = fix2float(dut.o_output_13)
    layer2[count, count2, 14] = fix2float(dut.o_output_14)
    layer2[count, count2, 15] = fix2float(dut.o_output_15)
    layer2[count, count2, 16] = fix2float(dut.o_output_16)
    layer2[count, count2, 17] = fix2float(dut.o_output_17)
    layer2[count, count2, 18] = fix2float(dut.o_output_18)
    layer2[count, count2, 19] = fix2float(dut.o_output_19)
    
    return layer2

I get the error that new_x is not defined. If I change the arguments new_x=new_x to just new_x and follow the same for other arguments, the calling function's new_x is assigned to dut. 

In any example, I have not seen cocotb async function with dut being called. But I want to know if what I am doing is possible in cocotb. If so, let me know where I am going wrong.

Thanks.

ktbarrett, 2021-01-04:
I see in your Python script you are using asyncio to run the `test_cnn_hw` coroutine. Is that the cocotb test you posted or something else? cocotb is not compatible with asyncio. cocotb tests are where all tests *must* start. So you will have to run your Python NN model *in* the cocotb test.

kgokarn, 2021-01-06:
Thank you. Is async function mandatory to send stimuli to hardware in cocotb or will synchronous function work as well?

ktbarrett, 2021-01-07:
> Is async function mandatory to send stimuli to hardware in cocotb or will synchronous function work as well?

cocotb tests must be async functions.

> If I do use async function, then my loop counter increments to its cap and iterator that needs to be sent to the hardware is sent only with the highest number instead for all.

You cannot create multiple cocotb tests in a loop in that way. Just move this logic into the test body.

```python
@cocotb.test()
async def hw_stimuli(dut):
    for i in range(15):
        dut.abc <= x[i]
```

kgokarn, 2021-01-13:
Hi, I still have errors with this code, although I am running the complete neural network within the async function. When I read the output in testbench, I get XXXXXXXXXXXX. 

pi_fxp = Fxp(None, signed=True, n_word=12, n_frac=11)    
pi_fxp.rounding = 'around'   

def fix_pt(val):                          
    return int(pi_fxp(val).bin(),2)

def fix2float(val):
    f2f= fix_to_float(True,12, 11)
    return float(f2f(int(val)))

@cocotb.test()
async def test_cnn_hw(dut, cnn_data=cnn_data, new_x=new_x, layer2=layer2):
    clock = Clock(dut.clk, 10, units="ns")  # Create a 10ns period clock on port clk
    cocotb.fork(clock.start())  # Start the clock
    reset = 1
    for i, img in enumerate(cnn_data['x_test']):

        # layer0: conv2d
        weights = cnn_data['l0weights']
        iter=0
        count = 0
        while (iter < 25):
            ii = 0
            count2 = 0
            while (ii < 25):
                new_x = img[ii:ii + 5, iter:iter + 5]
                
                
                await RisingEdge(dut.clk)
                dut.reset <= reset
                dut.i_activation_0_0 <= fix_pt(new_x[0, 0])
                dut.i_activation_0_1 <= fix_pt(new_x[0, 1])
                dut.i_activation_0_2 <= fix_pt(new_x[0, 2])
                dut.i_activation_0_3 <= fix_pt(new_x[0, 3])
                dut.i_activation_0_4 <= fix_pt(new_x[0, 4])
                dut.i_activation_1_0 <= fix_pt(new_x[1, 0])
                dut.i_activation_1_1 <= fix_pt(new_x[1, 1])
                dut.i_activation_1_2 <= fix_pt(new_x[1, 2])
                dut.i_activation_1_3 <= fix_pt(new_x[1, 3])
                dut.i_activation_1_4 <= fix_pt(new_x[1, 4])
                dut.i_activation_2_0 <= fix_pt(new_x[2, 0])
                dut.i_activation_2_1 <= fix_pt(new_x[2, 1])
                dut.i_activation_2_2 <= fix_pt(new_x[2, 2])
                dut.i_activation_2_3 <= fix_pt(new_x[2, 3])
                dut.i_activation_2_4 <= fix_pt(new_x[2, 4])
                dut.i_activation_3_0 <= fix_pt(new_x[3, 0])
                dut.i_activation_3_1 <= fix_pt(new_x[3, 1])
                dut.i_activation_3_2 <= fix_pt(new_x[3, 2])
                dut.i_activation_3_3 <= fix_pt(new_x[3, 3])
                dut.i_activation_3_4 <= fix_pt(new_x[3, 4])
                dut.i_activation_4_0 <= fix_pt(new_x[4, 0])
                dut.i_activation_4_1 <= fix_pt(new_x[4, 1])
                dut.i_activation_4_2 <= fix_pt(new_x[4, 2])
                dut.i_activation_4_3 <= fix_pt(new_x[4, 3])
                dut.i_activation_4_4 <= fix_pt(new_x[4, 4])

                dut.i_weight_0_0_0 <= fix_pt(weights[0, 0, 0])
                dut.i_weight_0_0_1 <= fix_pt(weights[0, 1, 0])
                dut.i_weight_0_0_2 <= fix_pt(weights[0, 2, 0])
                dut.i_weight_0_0_3 <= fix_pt(weights[0, 3, 0])
                dut.i_weight_0_1_0 <= fix_pt(weights[1, 0, 0])
                dut.i_weight_0_1_1 <= fix_pt(weights[1, 1, 0])
                dut.i_weight_0_1_2 <= fix_pt(weights[1, 2, 0])
                dut.i_weight_0_1_3 <= fix_pt(weights[1, 3, 0])
                dut.i_weight_0_2_0 <= fix_pt(weights[2, 0, 0])
                dut.i_weight_0_2_1 <= fix_pt(weights[2, 1, 0])
                dut.i_weight_0_2_2 <= fix_pt(weights[2, 2, 0])
                dut.i_weight_0_2_3 <= fix_pt(weights[2, 3, 0])
[... 341 lines cut ...]
                layer2[count, count2, 9] = fix2float(dut.o_output_9)
                layer2[count, count2, 10] = fix2float(dut.o_output_10)
                layer2[count, count2, 11] = fix2float(dut.o_output_11)
                layer2[count, count2, 12] = fix2float(dut.o_output_12)
                layer2[count, count2, 13] = fix2float(dut.o_output_13)
                layer2[count, count2, 14] = fix2float(dut.o_output_14)
                layer2[count, count2, 15] = fix2float(dut.o_output_15)
                layer2[count, count2, 16] = fix2float(dut.o_output_16)
                layer2[count, count2, 17] = fix2float(dut.o_output_17)
                layer2[count, count2, 18] = fix2float(dut.o_output_18)
                layer2[count, count2, 19] = fix2float(dut.o_output_19)

                ii = ii+2
                count2 = count2+1

            iter = iter+1
            count = count+1

The error message is: raise ValueError("Unable to resolve to binary >%s<" % string)
                                                                                                                     ValueError: Unable to resolve to binary >xxxxxxxxxxxx<
     0.00ns ERROR    cocotb.regression                         regression.py:476  in _log_test_summary               Failed 1 out of 1 tests (0 skipped)

The hardware is validated. Also when I send stimuli through cocotb, the control is passed to Verilog. Do you think, something that I have done here is wrong?

Thanks.

eric-wieser, 2021-01-13:
Your code is very hard to follow because of the large number of assignments. Can I suggest you replace
```python
dut.i_activation_0_0 <= fix_pt(new_x[0, 0])
dut.i_activation_0_1 <= fix_pt(new_x[0, 1])
...
```
with
```
for i in range(n):
    for j in range(m):
        getattr(dut, f"i_activation_{i}_{j}") <= fix_pt(new_x[i, j])
            dut.i_activation_0_1 <= fix_pt(new_x[0, 1])
```

kgokarn, 2021-02-15:
Sorry for the late reply. I had my thesis submission, therefore I didn't check.

I have modified the code to better readability. But the error persists.

pi_fxp = Fxp(None, signed=True, n_word=12, n_frac=11)    
pi_fxp.rounding = 'around'   

def fix_pt(val):                          
    return int(pi_fxp(val).bin(),2)

def fix2float(val):
    f2f= fix_to_float(True,12, 11)
    return float(f2f(int(val)))

# for each input in the test set
@cocotb.test()
async def test_cnn_hw(dut, cnn_data=cnn_data):
    clock = Clock(dut.clk, 10, units="ns")  # Create a 10ns period clock on port clk
    cocotb.fork(clock.start())  # Start the clock
    reset = 1
    for i, img in enumerate(cnn_data['x_test']):

        # layer0: conv2d
        weights = cnn_data['l0weights']
        iter=0
        count = 0
        while (iter < 25):
            ii = 0
            count2 = 0
            while (ii < 25):
                new_x = img[ii:ii + 5, iter:iter + 5]
                
                
                await RisingEdge(dut.clk)
                dut.reset <= reset
                for i2 in range(5):
                    for i3 in range(5):
                        getattr(dut, f"i_activation_{i2}_{i3}") <= fix_pt(new_x[i2, i3])

                for i4 in range(20):
                    for i5 in range(4):
                        for i6 in range(4):
                            getattr(dut, f"i_weight_{i4}_{i5}_{i6}") <= fix_pt(weights[i5,i6,i4])

                time.sleep(20)
                
                

                layer2[count, count2, 0] = fix2float(dut.o_output_0)
                layer2[count, count2, 1] = fix2float(dut.o_output_1)
                layer2[count, count2, 2] = fix2float(dut.o_output_2)
                layer2[count, count2, 3] = fix2float(dut.o_output_3)
                layer2[count, count2, 4] = fix2float(dut.o_output_4)
                layer2[count, count2, 5] = fix2float(dut.o_output_5)
                layer2[count, count2, 6] = fix2float(dut.o_output_6)
                layer2[count, count2, 7] = fix2float(dut.o_output_7)
                layer2[count, count2, 8] = fix2float(dut.o_output_8)
                layer2[count, count2, 9] = fix2float(dut.o_output_9)
                layer2[count, count2, 10] = fix2float(dut.o_output_10)
                layer2[count, count2, 11] = fix2float(dut.o_output_11)
                layer2[count, count2, 12] = fix2float(dut.o_output_12)
                layer2[count, count2, 13] = fix2float(dut.o_output_13)
                layer2[count, count2, 14] = fix2float(dut.o_output_14)
                layer2[count, count2, 15] = fix2float(dut.o_output_15)
                layer2[count, count2, 16] = fix2float(dut.o_output_16)
                layer2[count, count2, 17] = fix2float(dut.o_output_17)
                layer2[count, count2, 18] = fix2float(dut.o_output_18)
                layer2[count, count2, 19] = fix2float(dut.o_output_19)

                ii = ii+2
                count2 = count2+1

            iter = iter+1
            count = count+1

When I read the output from the testbench, I read only don't cares.

eric-wieser, 2021-02-15:
When using hardware, you have to remember that there are two entirely different clocks: the wall time, and the simulator time.
While cocotb is running, the simulator is paused. When you do `time.sleep(20)`, the simulator remains paused (0s of simulator tmie passes), and nothing happens for 20s of wall time.

If you want the simulator to resume, you have to use `await Timer(20, units="ms")` or similar, which advances 20ms of _simulator_ time (and an unknown amount of wall time passes).

kgokarn, 2021-02-16:
Thank you for your comment, I will change that and check. Thanks again.

kgokarn, 2021-02-17:
I have replaced "time.sleep(20)" with "await Timer(20000000, units="ns")", however, I still have the same problem.

kgokarn, 2021-02-17:
This problem still persists even when "time.sleep()" is replaced by "await Timer(duration, units="ns").

eric-wieser, 2021-02-17:
from a deleted comment

> I have replaced "time.sleep(20)" with "await Timer(20000000, units="ns")", however, I still have the same problem

It's highly unlikely you want to use such a long duration, since you would have to wait for the simulator to simulator 2 million clock cycles. Are you sure you don't intend to wait for clock cycles instead?

kgokarn, 2021-02-17:
My clock speed in the design is 20ns. I do have many state machines in the hardware. Precisely, 2520 state machines. I think I can wait for 2520 clock cycles. I kept higher simulator time to be safe.

Do I wait for 2520 clock cycles? I do see await on the rising edge of the clock. For 2520 clock cycles, should I run the await statement on the rising edge in the loop, or are there are alternate ways?

eric-wieser, 2021-02-17:
`await`ing in a loop is fine. You can use `ClockCycles` instead, but all that does is the loop anyway.

kgokarn, 2021-02-17:
Yes, I have await on the rising edge in a for loop for the specified number of cycles. I read don't care from the hardware. Could this be a problem from something else?

Let me know if I need to produce any further details.

eric-wieser, 2021-02-17:
This could be a problem in your HDL. Can you try reading another signal from your design and check that it does not contain "don't care"?

kgokarn, 2021-02-17:
I had an additional output port defined to check. I was getting don't care on that as well (it is assigned '1' in the final state). However, I will check the HDL code completely once again and let you know.

Thanks.

cmarqu, 2021-02-17:
Are you sure your reset works, and that you get out of reset?

kgokarn, 2021-02-21:
Hi, thank you for your comments. Yes, I had to reset the network once. Now the don't care problem has been resolved.

kgokarn, 2021-02-22:
The issue has been resolved. It was indeed caused by not resetting the hardware.  You may close the issue.
Thanks again for your support.

closed 2021-02-22 by cmarqu