cocotb/cocotb #2326: SystemVerilog events are not supported

cmarqu, 2021-01-05:
In cocotb 1.5.0dev0, we do not support SystemVerilog events (`event myevent; ->myevent;`):

```
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:301  in __gpi_get_handle_by_name        Searching for myevent
     0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:314  in __gpi_get_handle_by_name        Checking if myevent is native through implementation VPI
     0.00ns WARNING  cocotb.gpi                                  VpiImpl.cpp:260  in create_gpi_obj_from_handle      VPI: Not able to map type vpiNamedEvent(34) to object.
     0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:371  in native_check_create             Unable to fetch object myevent
```

ktbarrett, 2021-01-05:
How would you suggest this be supported? Does the SV VPI extensions have methods for operating on events? Since this is a SV-specific feature, and there is no parallel in VHDL, should this be supported in an extension? I think either way we need to discuss how we plan on supporting language/interface-specific features in the GPI (if at all) going forward.

cmarqu, 2021-01-05:
I intuitively tried `Edge(myevent)` so that could be the UI for reacting on events originating on the HDL side.
The other way 'round (i.e. triggering an  event from Python) I'm not sure.