cocotb/cocotb #2328: Cadence Xcelium breaks escape sequence when reading ANSI-colored strings via VPI

cmarqu, 2021-01-05:
When trying to read a SystemVerilog string with ANSI color codes in it, Xcelium (Version 20.09) strips the escape code (, \x1b) and thus breaks the coloring.

Writing the same string toward SystemVerilog works fine; doing this on a VHDL design works fine for both writing and reading.
(Questa and VCS also work fine with SystemVerilog for both writing and reading.)

This seems to be a simulator bug that cocotb can't do anything about.

cmarqu, 2021-01-05:
This issue has been opened for documentation purposes only, closing it since cocotb is helpless here.

closed 2021-01-05 by cmarqu

garmin-mjames, 2021-01-05:
I wonder if printing log messages with `vpi_printf()` would work.