cocotb/cocotb #2332: FLI library problem when running cocotb on linux debian buster

xlz-sbarillet, 2021-01-07:
Hello,
I have the same behavior than #2277 ticket

My configuration is:

- the cocotb version used is 1.4.0, then manually updated to 1.5.0-dev0. (both have the same behavior)
- the operatinfgsystem is a debian buster 64 bit
- the simulator is questasim 2020.4_2 64bit only (and GCC 64bit installed)
- the Python version: Python 3.7.
- ZSH shell env (oh my zsh)
- PATH env contains questasim/bin path.
 A reduced $PATH env contains :
/opt/cao/tools/vendor/mentor/questasim_2020.4_2/questasim/gcc-7.4.0-linux_x86_64/bin:/opt/cao/tools/vendor/mentor/questasim_2020.4_2/questasim/bin:/home/sbarillet/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/snap/bin

Both cocotb version generates the same files into: ls -al /usr/local/lib/python3.7/dist-packages/cocotb/libs                                                                         libcocotb.so libcocotbutils.so libcocotbvhpi_aldec.so libcocotbvhpi_ius.so libcocotbvpi_aldec.so libcocotbvpi_ghdl.so libcocotbvpi_icarus.vpl libcocotbvpi_ius.so libcocotbvpi_modelsim.so libcocotbvpi_vcs.so libcocotbvpi_verilator.so libembed.so libgpilog.so libgpi.so libpygpilog.so
Obviously, i didn't found the "libcocotbfli_modelsim.so" requested during simulation, and I try to figure out why this FLI libraries was not compiled.
if I execute: ls -l $(dirname $(which vsim))/../include/ | grep mti , 4 files was found:
 mti.h  mtiq.h  mti_traverse.h  mti_vpi_user.h

0. I double-check the Questasim Path and force python installation as requested:  python -m pip install --force-reinstall git+https://github.com/cocotb/cocotb.git

1. I download cocotb examples from: git clone https://github.com/potentialventures/cocotb.git

2. I choose to run the "adder" example by typing the following command from anaconda Prompt:

make SIM=questa TOPLEVEL_LANG=vhdl
make results.xml
make[1]: Entering directory '/home/sbarillet/Documents/cocotb/cocotb/examples/adder/tests'
ERROR: cocotb was not installed with an FLI library, as the mti.h header could not be located.
If you installed an FLI-capable simulator after cocotb, you will need to reinstall cocotb.
Please check the cocotb documentation on ModelSim support.
make[1]: *** [/usr/local/lib/python3.7/dist-packages/cocotb/share/makefiles/simulators/Makefile.questa:75: /usr/local/lib/python3.7/dist-packages/cocotb/libs/libcocotbfli_modelsim.so] Error 1
make[1]: Leaving directory '/home/sbarillet/Documents/cocotb/cocotb/examples/adder/tests'
make: *** [/usr/local/lib/python3.7/dist-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2

Can you please help me to fix this issue ?
Thanks for your help.

ktbarrett, 2021-01-07:
Is `vdbg` in your PATH? `vdbg` is used rather than `vsim` to detect if questa is present.

themperek, 2021-01-07:
https://docs.cocotb.org/en/stable/simulator_support.html#mentor-modelsim

xlz-sbarillet, 2021-01-08:
Forgot to mention but I've checked that vdbg was in the PATH with following command:

- type -p "vsim" && echo "in path" || echo "not in path"                                                                                     ✔  at 09:03:58 
vsim is /opt/cao/tools/vendor/mentor/questasim_2020.4_2/questasim/bin/vsim
**in path**

- type -p "vdbg" && echo "in path" || echo "not in path"                                                                                    ✔  at 09:04:17   
vdbg is /opt/cao/tools/vendor/mentor/questasim_2020.4_2/questasim/bin/vdbg
**in path**

themperek, 2021-01-08:
You have `vdbg` in PATH during cocotb installation?

xlz-sbarillet, 2021-01-08:
Yes, `vdbg` executable was in the PATH during cocotb installation.

themperek, 2021-01-08:
Please provide log file from `pip install -v --force-reinstall cocotb`

xlz-sbarillet, 2021-01-08:
I finally found the problem. it came from the vdbg executable that was not in the PATH as mentioned. I was checking the PATH environment variable under my user session while installing the tool as root and the root PATH had an incorrect path to Questa.
Thank you again for your help.

closed 2021-01-08 by xlz-sbarillet

ktbarrett, 2021-01-08:
I might be misinterpreting this.

>while installing the tool as root 

You should not install packages as root. If your site-packages directory is not writable by your user, and your version of pip is old enough if doesn't catch this case, explicitly install the package as a user by adding the `-U` flag to the `pip` command.