cocotb/cocotb #2344: [Riviera-PRO] vpi_register_cb returns reused callback handle

garmin-mjames, 2021-01-13:
VSimSA is re-using callbacks for `cbValueChange`.

In `test_compare` you have this pattern:
```py3
    await RisingEdge(dut.clk)
    ...
    await FallingEdge(dut.clk)
```
So, we have a `cbValueChange` callback handle for the `dut.clk` signal, and after the `RisingEdge` trigger fires, we are inside that callback.
Then we yield `FallingEdge`, which when primed will register for a new `cbValueChange` callback on the same signal.
Riviera gives us back the same handle we already have for `RisingEdge` (but doesn't update the callback data for the new `FallingEdge` object).
We then cleanup `RisingEdge`, which calls `vpi_remove_cb` on the shared handle, sets the handle to NULL, and sets the state to `GPI_FREE`.
On the falling edge of `dut.clk`, the callback fires again but with the original `RisingEdge` callback object that is `GPI_FREE` so nothing happens, it just returns to the simulator.

I'm pretty sure this falls afoul of the VPI standard. I found this, which is in **38.36.1.1** in context of callbacks on statements:

> Multiple calls to `vpi_register_cb()` with the same data shall result in multiple callbacks.

This problem didn't happen before because the handle for `VpiValueCbHdl` objects was always removed by the scheduler with `trigger.unprime`.

_Originally posted by @garmin-mjames in https://github.com/cocotb/cocotb/pull/2338#issuecomment-759778275_

imphil, 2022-04-23:
Filed a support case with Aldec, case number SPT81300.

imphil, 2023-10-13:
The 2023.04 release notes mention this issue is resolved.

@marlonjames Is there something we want to do as result of that on the cocotb side?

marlonjames, 2023-10-13:
`test_both_edge_triggers` should now pass with riviera 2023.04:
https://github.com/cocotb/cocotb/blob/c34c2bc5cf2408c0c5c386d5d9c139ecf6453889/tests/test_cases/test_cocotb/test_edge_triggers.py#L232-L244

imphil, 2023-11-01:
I confirmed that this issue is fixed with Riviera 2023.10. https://github.com/cocotb/cocotb/pull/3471 removes the test exclusion.

closed 2023-11-01 by imphil