cocotb/cocotb #2346: [Verilator] Can not find root handle

garmin-mjames, 2021-01-14:
cocotb: master ad1fe4883b2fc6f7
os: Ubuntu 20.04 (GitHub Actions)
simulator: Verilator version 4.109 devel
Python: 3.8.6

---

Can be seen in GitHub Actions:
https://github.com/cocotb/cocotb/runs/1697054347?check_suite_focus=true#step:22:141

This recent Verilator PR looks like the likely culprit: https://github.com/verilator/verilator/pull/2704

Currently cocotb uses the `--prefix Vtop` option for Verilator to ensure the name is always the same, regardless of `TOPLEVEL`. This allows hard-coding of `#include "Vtop.h"` in `verilator.cpp`.
We also use a "nameless" top level in `verilator.cpp`, which was causing some problems with getting the right handles to the top level ports (see #2244).

I think that these things are interacting in an unexpected way.

Here is the output from `Verilated::internalsDump()` for `sample_module`:
```
internalsDump:
  Version: Verilator 4.109 devel
  Argv: sim_build/sample_module +verilator+debug
  scopesDump:
    SCOPE 0x55fa67817e98: TOP
       VAR 0x55fa67818058: and_output
       VAR 0x55fa678180d8: clk
       VAR 0x55fa67818158: inout_if
       VAR 0x55fa678181d8: stream_in_data
       VAR 0x55fa67818258: stream_in_data_39bit
       VAR 0x55fa678182d8: stream_in_data_dqword
       VAR 0x55fa67818358: stream_in_data_dword
       VAR 0x55fa678183d8: stream_in_data_wide
       VAR 0x55fa67818458: stream_in_int
       VAR 0x55fa678184d8: stream_in_ready
       VAR 0x55fa67818558: stream_in_real
       VAR 0x55fa678185d8: stream_in_string
       VAR 0x55fa67818658: stream_in_valid
       VAR 0x55fa678186d8: stream_out_data_comb
       VAR 0x55fa67818758: stream_out_data_registered
       VAR 0x55fa678187d8: stream_out_int
       VAR 0x55fa67818858: stream_out_ready
       VAR 0x55fa678188d8: stream_out_real
    SCOPE 0x55fa67817ed0: sample_module
       VAR 0x55fa67818998: STRING_CONST
       VAR 0x55fa67818a18: STRING_VAR
       VAR 0x55fa67818a98: _underscore_name
       VAR 0x55fa67818b18: and_output
       VAR 0x55fa67818b98: array_0_to_3
       VAR 0x55fa67818c18: array_3_downto_0
       VAR 0x55fa67818c98: array_4_to_7
       VAR 0x55fa67818d18: array_7_downto_4
       VAR 0x55fa67818d98: clk
       VAR 0x55fa67818e18: inout_if
       VAR 0x55fa67818e98: logic_a
       VAR 0x55fa67818f18: logic_b
       VAR 0x55fa67818f98: logic_c
       VAR 0x55fa67819018: mybit
       VAR 0x55fa67819098: mybits
       VAR 0x55fa67819118: mybits_uninitialized
       VAR 0x55fa67819198: register_array
       VAR 0x55fa67819218: stream_in_data
       VAR 0x55fa67819298: stream_in_data_39bit
       VAR 0x55fa67819318: stream_in_data_dqword
       VAR 0x55fa67819398: stream_in_data_dword
       VAR 0x55fa67819418: stream_in_data_wide
       VAR 0x55fa67819498: stream_in_int
       VAR 0x55fa67819518: stream_in_ready
       VAR 0x55fa67819598: stream_in_real
       VAR 0x55fa67819618: stream_in_string
       VAR 0x55fa67819698: stream_in_valid
       VAR 0x55fa67819718: stream_out_data_comb
       VAR 0x55fa67819798: stream_out_data_registered
       VAR 0x55fa67819818: stream_out_int
       VAR 0x55fa67819898: stream_out_ready
       VAR 0x55fa67819918: stream_out_real
       VAR 0x55fa67819998: struct_var
       VAR 0x55fa67819a18: temp
```

We would like to interact through VPI with the `sample_module` module by getting its handle with `vpi_iterate(vpiModule, NULL)`, and to get top-level ports we can read/write from that handle or with calls like `vpi_handle_by_name('sample_module.clk', NULL)`.

@wsnyder are you able to provide some guidance on how we should structure things?

wsnyder, 2021-01-14:
The --prefix sets the filenames, but is not intended to affect the VPI scopes.  The name "TOP" comes from the name used when you new() the Verilated model.  It's possible passing "" when you construct the model will do what you want, but it's also possible that some code changes would be needed to make that work.

garmin-mjames, 2021-01-14:
@wsnyder We are passing `""` already:

https://github.com/cocotb/cocotb/blob/ad1fe4883b2fc6f75682651fa31c53d161414a29/cocotb/share/lib/verilator/verilator.cpp#L54

For the `t_vpi_module` Verilator test, `Vt_vpi_module__Syms.cpp` sets up the scopes and hierarchy like so:
```cpp
// Setup scopes
    __Vscope_t.configure(this, name(), "t", "t", -12, VerilatedScope::SCOPE_MODULE);
    __Vscope_t__mod_a.configure(this, name(), "t.mod_a", "mod_a", -12, VerilatedScope::SCOPE_MODULE);
    __Vscope_t__mod_a__mod_c__02E.configure(this, name(), "t.mod_a.mod_c__02E", "mod_c.", -12, VerilatedScope::SCOPE_MODULE);
    
    // Set up scope hierarchy
    __Vhier.add(0, &__Vscope_t);
    __Vhier.add(&__Vscope_t, &__Vscope_t__mod_a);
    __Vhier.add(&__Vscope_t__mod_a, &__Vscope_t__mod_a__mod_c__02E);
```

But for cocotb's `sample_module`, in `Vtop__Syms.cpp` we get:
```cpp
    // Setup scopes
    __Vscope_TOP.configure(this, name(), "TOP", "TOP", 0, VerilatedScope::SCOPE_OTHER);
    __Vscope_sample_module.configure(this, name(), "sample_module", "sample_module", -12, VerilatedScope::SCOPE_OTHER);
    
    // Set up scope hierarchy
    
```
The extra `TOP` scope may or may not be relevant, but the `sample_module` scope is `VerilatedScope::SCOPE_OTHER` instead of `VerilatedScope::SCOPE_MODULE`, which I assume is why we are unable to get a handle to it through VPI.

garmin-mjames, 2021-01-20:
Somehow the `$display` blocks added to `sample_module.sv` in #2322 are causing this.

https://github.com/cocotb/cocotb/blob/5d4ae7af2aa357dbb7d12fcb078732fbf40c4bfc/tests/designs/sample_module/sample_module.sv#L158-L166

If I remove them, everything works.

marlonjames, 2024-01-03:
This appears to be back, shows up in `test_fatal`. See #3237

ktbarrett, 2026-04-18:
I think this is done. Re can reopen if not.

PR #2363 (2021-01-21): Remove sample_module calls to $display when using Verilator
Closes #2346.

When the always blocks that display value changes to the bit variables are included,
Verilator is unable to find a handle to the top-level module through VPI.

There is some underlying issue in Verilator that I haven't figured out, but this is a work-around for now.
closed 2026-04-18 by ktbarrett