cocotb/cocotb #2350: Ability to access nested structures without getattr [Quality of Life]

jshaker000, 2021-01-16:
This isnt a very important improvement, but I think it would a nice QoL improvement or a nice thing to explicitly doc.

But if I have top level module `T` that instatiates module `bar` as `inst_bar`, then to access signals in bar I need to use getattr on T, rather then perhaps naturally going

T.inst_bar.<>

From a datastructures perspective, its perhaps nicer to pass around a handle to a point in a design (for say attaching a score board or injection point hierarchically), rather than essentially needing to pass around strings.

If this isnt so easy to add, I think maybe doc'ing this more explicitly would be helpful, because it definitely took me by surprise and there was a lot of head scratching until I opened the source code of some of the provided Bus's / Monitors as examples.

eric-wieser, 2021-01-16:
I don't understand what you're asking for here. Can you give a code snippet showing what you do today that you find awkward?

jshaker000, 2021-01-19:
To be clearer, say that I have a top level module T, with child modules A and B. To refer to a net in A, I need to do

getattr(dut, "A.<net_name>")

Rather than hierarchically going

dut.A.<net_name>

Presumably this is because dut has attribute "A.<net_name>", but "A" does not have attribute "<net_name>

Does that help at all? If not I can come up with a simple example with some verilog and some python.

ktbarrett, 2021-01-19:
If you mean that inside of an *instance* of module T, you have an *instance* of module A, this absolutely should be possible and may be a bug in the simulator/cocotb support of your simulator. If you mean that you want to access a signal in a module *definition*, that will probably never be supported since I doubt the VPI and VHPI support accessing module definitions.

jshaker000, 2021-01-19:
I was able to figure this out, I'm not sure why it wasnt working for me before.

closed 2021-01-19 by jshaker000

ktbarrett, 2021-01-19:
Simulator support changes with the wind. Glad it works now.