cocotb/cocotb #2351: Expend cocotb-config to return path to vpi/fli/vhpi library

themperek, 2021-01-16:
Considering that https://github.com/cocotb/cocotb/pull/1798 will make vpi/fli/vhpi different depending on platform/os we can add extra functionality to `cocotb-config` that would allow easier custom flow integration.

An example:
`cocotb-config --vpi-lib icarus` return absolut path to `libcocotbvpi_icarus.so`
`cocotb-config --vpi-lib questa` return absolut path to `libcocotbvpi_modelsim.so`
`cocotb-config --lib-dir` return path where `libcocotbvpi_icarus.so` is located

Could also add something like this:
`cocotb-config --vpi-cmd icarus` that would return `-M $(cocotb-config --prefix)/cocotb/libs -m libcocotbvpi_icarus`
`cocotb-config --fli-cmd questa` that would return `-foreign "cocotb_init $(cocotb-config --prefix)/cocotb/libs/libcocotbfli_modelsim.so`

Any support for this?
Any suggestions?

Related https://github.com/cocotb/cocotb/pull/2340

ktbarrett, 2021-01-16:
I agree that something like this would be useful. At a minimum giving the directory of the libs and the name of each particular lib will be useful. Especially if we do any renaming or combining of libraries in the future, creating this abstraction should allow those changes to not break users. It might also be useful for people making GPI extensions

I'm thinking
```
cocotb-config --includes                          # prints to include dirs for VPI, VHPI, FLI, and GPI (for use with '-I$(cocotb-config --includes)')
cocotb-config --lib-dirs                          # prints directory containing libgpi (for use with '-L$(cocotb-config --lib-dirs)')
cocotb-config --lib-path {interface} {simulator}  # prints absolute path to library the GPI implementation of a given interface for a simulator  (for use with custom flows)
cocotb-config --cmd-args {interface} {simulator}  # prints arguments to pass simulator to get it to run cocotb (for use with custom flows)
```

cmarqu, 2021-01-16:
`--vpi-cmd` or `--cmd-args` would need an additional `--format` option because not all flows will require the `$(...)` syntax for calling a command, but in general, I like this idea.

themperek, 2021-01-17:
> `--vpi-cmd` or `--cmd-args` would need an additional `--format` option because not all flows will require the `$(...)` syntax for calling a command, but in general, I like this idea.

Maybe we can return raw text without parameters (`$(...)` syntax) in this case?

eric-wieser, 2021-01-17:
Even returning raw text is tricky, because posix and windows have different conventions for argument quoting - so you'd want separate `--sh-cmd-args` and `--batch-cmd-args` or similar

themperek, 2021-01-17:
I will try to do the first part after #1798 and then will see if/how to handle `--cmd-args`.

ktbarrett, 2021-02-10:
Did we decide that implementing `--cmd-args` is going to be rather difficult and fragile? If so, the current addition of `--lib-path` closes this issue.

themperek, 2021-02-10:
> Did we decide that implementing `--cmd-args` is going to be rather difficult and fragile? If so, the current addition of `--lib-path` closes this issue.

I am at the moment not convinced if this will improve anything. We can give/point to an example. It is part of simulator documentation.

closed 2021-07-22 by ktbarrett