cocotb/cocotb #2353: stale bot no longer closing stale issues?

ktbarrett, 2021-01-17:
The action is running, but recently it's not been closing issues it has marked as stale (see dates on #2074). I wonder if there are questions it is also not marking as stale?

ktbarrett, 2021-01-20:
Some discussion in https://github.com/actions/stale/issues/284. It *might* be fixed in the next release, but seeing as I don't get an error message, I'm not so sure that's the same issue.

imphil, 2021-01-20:
ACTIONS_STEP_DEBUG=true is set as secret in the cocotb repository.

ktbarrett, 2021-03-03:
Development is paused until we release (~1 week from now). I'll bump the `operations-per-run` and maybe remove the `close?` tag after that.

C0ZEN, 2021-03-03:
You have this in your configuration `only-labels: type:question,status:close?`.
Are you aware that it's working only if both labels are on the issue/PR at the same time?
Also `operations-per-run: 5` should be set to `operations-per-run: 30` IMO so that you are sure to handle more issues and PR per run.
Lastly, the next stale release will include this https://github.com/actions/stale/issues/308 which will allow you to only handle issue/PR with either `type:question` or `status:close?` labels if set like this `any-of-labels: type:question,status:close?` instead of the current `only-labels` one.

ktbarrett, 2021-03-03:
Ahh, I was not aware that `only-labels` was an intersection, it looks like the documentation was just updated to indicate that. I'll switch to `any-of-labels`. Also, [there is a duplicate](https://github.com/actions/stale/blame/main/README.md#L26-L27).

C0ZEN, 2021-03-03:
@ktbarrett if you want to go ahead of schedule you can target the main branch nevertheless if you want to play it safe and let the new version be more tested in a real repository you can wait for the next release.
Nice catch regarding the documentation and it's already fixed here https://github.com/actions/stale/pull/341 😎

PR #2446 (2021-03-06): Fix stale bot
Closes #2353. The tag `only-labels` implies only an intersection of the listed labels counts for marking as stale, so we switch to `any-of-labels`. Also bumped `operations-per-run` up to 30 to help mark more each day.
closed 2021-03-06 by imphil